TomWhitwell/TuringMachine #104: No LEDs or output

ihgrant, 2018-09-04:
The unit doesn't respond at all when I power it up - no LEDs, no output. Here's what I've done so far:

- I checked my power header for shorts (none)
- I measured the voltage across the back board's TL072 (24V, which I understand is good)
- I checked the soldering on both sides of the the 6-pin connector between the two boards (seems ok to me).

For some background, and in case it matters, when I initially finished the build and plugged it in, the LEDs turned on and something happened when I plugged in a clock, but the light for the -12V rail on my power supply stopped lighting up (worked without the Turing Machine plugged in). From reading some other issues here I figured out that I got one of the diodes backwards, and fixed that mistake. Now the -12V rail is good when the TM is plugged in.

[image]

[image]

[image]

[image]

Thonk-Steve, 2018-09-04:
Hi Ian, are you getting anything out of the noise output when it's plugged
in? Was it one of the diodes next to the power header that was positioned
incorrectly on first power up?

Steve

ihgrant, 2018-09-05:
I do hear noise from the noise output, and yes, the top 1N4001 diode in my photo above was the one that was backwards.

ihgrant, 2018-09-09:
Messing around with it a bit more this weekend I've noticed that if I power it on without the panel on, all of the top LEDs turn on, and with a square audio wave into the Clock, at some positions of the Scale and Change knobs, I can get some noise from the output.

If I put on the panel without securing any of the washers, a few of the LEDs will light up. If I secure the panel at all, or even just press it tight, none of the LEDs will light up.

Does this suggest a short with the panel, perhaps?

TomWhitwell, 2018-09-09:
Hi, My guess would be a missed solder connection somewhere - so a little flex in the board when the panel is attached might make the connection?

ihgrant, 2018-09-17:
Cleaned up some connections on the control board and the top LEDs light up with the panel on now.

If I were to start following traces with a continuity tester, where should I start?

ihgrant, 2018-11-11:
Really just recording this so that someone else might find my debugging process helpful, but if anyone has any advice I'm all ears.

Plugging in an oscillator's square wave to the clock worked after i routed it through a VCA and turned up the gain, which makes me think maybe I misplaced a resistor, so I'm checking all of the resistors on the control board.

Also when going through the trimming procedure I didn't notice any glitches in the audio output, or really any change at all while turning the Change knob, other than a lower frequency output at 12 to maybe 3 o'clock and higher frequency after that.

ihgrant, 2018-11-11:
Update: somehow I swapped the 10K resistors for 100K and vice-versa.

ihgrant, 2018-11-22:
Replacing the incorrect resistors with the correct ones has resulted in the clock being responsive, and I can get CV from Out now! I still have an issue regarding randomness (the lack thereof) but I haven't looked through the issues here to see what that might be yet. But it's alive!