TomWhitwell/TuringMachine #105: Replacement Electrolytic Capacitors

Techsture, 2018-09-09:
OK, new to this whole DIY thing and soldering for the most part.  Build a Chord Organ and Radio Music before trying to tackle the Turing Machine MKII.

Well, I still appear to have gotten ahead of myself.  While removing the electrolytic capacitors from the packaging, I yanked one of the legs off of one of them.  Being new to this, I shoved it back in thinking, "yeah, this is fine, I'm a pro now!"

Cut to a couple hours or so later and I'm plugging in my TM2.  The lights blip on real fast, then nothing.  Try again, not even a blip.

I checked a few things utilizing my limited knowledge of a multimeter, and I think the issue is the capacitor (one of them acts differently from the other).  Of course, took me about 30m of doing random stuff before I remembered the capacitor kerfuffle.

Anyways, what can I get for a replacement?  I think they're these, but I just want to be sure:

http://www.oddwires.com/20pcs-10uf-35v-85c-panasonic-eca-1vm100-electrolytic-capacitor/

Thanks, I love Music Thing Modular!

TomWhitwell, 2018-09-09:
Hi, sorry to hear that. Those replacement caps look OK to me
Be careful when you get them to check how they’ll fit - sometimes caps are slightly bigger than you expect and you have to work out a bit of a kludge to fit them in - for example leaving a bit of extra length on the ‘legs’ so you can lie them flat against the board. 
We specify extra compact ones for some kits to make it easier - so just check it all fits before you solder them in. But I’d definitely buy these because that retailer looks much more user friendly than Mouser. 

Also, of course, check & double check the polarity of the electro caps. 

There is always a risk with this kind of ‘magic smoke’ moment that something else will have been damaged in a short circuit, but fixing the cap definitely sounds like the place to start. 
Good luck!

closed 2018-09-09 by TomWhitwell

Techsture, 2018-09-12:
I just wanted to follow-up in case anyone in the future sees this.

The Panasonic 10uF 35V 85C caps DO work, I managed to get my Turing Machine MKII working!  👍 Just be aware that they are almost twice the height of the caps that come with the kit.  Leave a bit of room in the legs so you can tilt them slightly to the side so there's enough room to plug into the front board.

Thanks again @TomWhitwell for your help!