TomWhitwell/TuringMachine #106: Instructions/info on older version?

jasonnolan, 2018-10-16:
I just bought a kit and I think it is the old 2012 version. Definitely not the same layout and has a laser cut plexiglass cover. Are there online instructions for putting this together that anyone knows about? I can solder fine, but it is hard to be 100% clear about the components from what I've got. Thanks in advance.

Thonk-Steve, 2018-10-16:
Hey Jason,

Build doc here for the old Thonk kit -
https://www.thonk.co.uk/documents/turing/

Cheers

Steve

jasonnolan, 2018-10-16:
Hi Tom;

Thanks so much for the quick response and the instructions. 

BTW, I did also buy a MKII, but I'd like to get this MKI running as well.

Thanks for your time.

Jason

Dr. Jason Nolan
Assoc. Professor - Early Childhood Studies
Director - Responsive Ecologies Lab
Ryerson University

103N Bell Trinity Square
483 Bay Street
Toronto, Ontario
M5G2E1

jasonnolan, 2018-10-17:
Hi Steve;

I've got it. Downloaded it and it makes totally good sense. Also just have an MKII coming in and looking forward to having them.

best,

Jason