TomWhitwell/TuringMachine #107: Mk1 troubleshooting question

jasonnolan, 2018-11-17:
My Mk2 works great, but my Mk1, not so good. It worked for the first week or so, and then just fell apart. The ICs didn't fit (someone told me to jam them in, someone else told me to shave them down, tried both, no difference and no worse). When i patch in a clock, nothing happens with the lights. Scale seems fine (super fast pitches, but they do lower as I lower the knob) but when I put in a clock only the first led lights up, though when I set the dial to loop led 1 is on, 2 is off and rest are on. As I move the dial from loop to random this will suddenly stop. Noise comes out fine. I'm fine with soldering, but I'm not much on knowing how to run my own tests based on the schematics. Thanks

jasonnolan, 2018-11-22:
I've replaced all the ICs except the TL072CP. the device works for about 30 seconds. There's a continually blinking led in the first position, and there's randomness coming out for about 30 seconds and after that the first LED just blinks and nothing else until I flip the write lever up or down and then it runs for another 30 seconds, and goes back to just the first LED. Does that look like a TL072CP problem? Any other thoughts?

TomWhitwell, 2018-11-22:
Could 30 seconds cut out be a fundamental power issue - a short causing the fuses to blow - check if they (the yellow fuses that look a bit like capacitors) are hot to touch (careful!) when the module dies

jasonnolan, 2018-11-22:
Not hot at all. Thought they were caps! The TL074 is warm, but not hot. I'm going to make a video of it tomorrow and see if that help show what it is like. My MK2 works great, but since I built this one myself from the kit, I'm really eager go know what's wrong for the learning. Thanks for your response.

BTW, great article on pricing work/modules I saw I think on FB.

TomWhitwell, 2018-11-22:
Oh yes sorry, Mk1 doesn't have fuses - they're on the Backpack 

Schematic is here https://musicthing.co.uk/wp-content/uploads/2012/05/random_seq_schem_2-1_big.png

TL072 does the clock conditioning, so if the clock in works, it's OK.

jasonnolan, 2018-11-22:
Thanks. Looks like the same one in the build docs. When I get to the holiday break, I'll get more systematic and track down this weirdness.