TomWhitwell/TuringMachine #108: Turing 2 works but no lock, length does not seem to work

jss9h, 2019-01-01:
Dear Tom
Happy New Year!
First thanks for all the fantastic modules. Been using Turing 1 and expanders for quite a while and love them. Also Radio Music and Ears. 
I have built Turing 2 from the Thonk kit and it seems to work (it generates random bits, noise output is ok) but just keeps generating random bits and does not lock when the big knob is centred (or in any other position). Also what is almost certainly a related problem is that the length switch does not seem to have any effect and the random patterns remain random. (I realise that if the lock worked the length function may be ok).
I am an experienced kit builder and solderer. I have checked my soldering multiple times and suspect a duff chip or transistor. Can you please help?
Jonathan

jss9h, 2019-01-01:
> Hi,
> Sorry to hear you're having problems.
> I don't know if this was mistyped, but the lock should be at 5 o'clock on the big knob, not in the centre
> Until you have locking, the length switch is meaningless - a random stream is a random stream.
> Let me know if you get locking at 5 o'clock.
> Tom

Dear Tom
Wow that was a quick reply! I’ll try tonight.
Thanks again
Jonathan

Jonathan Shapero
From my iPad Air

jss9h, 2019-01-02:
Hello again Tom. I am slightly eggy-faced because I SHOULD have read the manual, but there is still a problem. Yes, it appears to lock at 5 o’clock but left running at about 1 Hz bits do occasionally change spontaneously -  about once every 2 minutes or so. I guess there is some calibration or adjustment required? Thanks again for your help. 

Jonathan 
From my iPad Air

TomWhitwell, 2019-01-02:
Hi, calibration details here: 
https://vimeo.com/163160088

closed 2019-01-02 by TomWhitwell

jss9h, 2019-01-02:
That looks straightforward enough!
Look forward to more great modules.
Many thanks

Jonathan
From my iPad Air