TomWhitwell/TuringMachine #109: Touring Maschine MKII after a while change the pattern in LOCK mode

TomyDeClerque, 2019-01-30:
Hi all ... my first post here :) 

so ... the problem I have is that after a while when I have TM2 in lock mode and have a pattern repeating ... something happen and it starts to outpu another pattern.

vpme.de Euclidean Circles are giving a clock to TM2 from here i go to Dooepfer Precision ADDER and in TipTop Quantizer and on VCO ...

when I get the " melody " that i like and start to mess with the sound ... sometimes it just changes the melody ... ( not always ) 

any help ? I bought TM2 already built ... is there anithing I can check ?

thanks

TomWhitwell, 2019-01-30:
Hi, best thing to start would be calibration: https://vimeo.com/163160088

mpsspsnl, 2019-08-06:
I may have encountered something similar. But I think the problem was caused by the quantizer which sometimes pulls the note up and sometimes down with the slightest voltage change.