TomWhitwell/TuringMachine #110: Turing machine CV not working

Djandy82, 2019-02-21:
I can't find other people whit this problem so i post it here. The cv function don't work and i calibrated it a few times. All the other functions work like it should and i can't find a solution to this cv problem. Mabye someone can guide me in the right direction.

Thonk-Steve, 2019-02-21:
Hi, do you mean the CV input?

Djandy82, 2019-02-21:
Yes the cv input. I checked the jack socket and it's not the case.

Thonk-Steve, 2019-02-21:
So my tip is try this: Turn the big knob fully anti-clockwise and then only
send it a positive DC voltage.
I don't have a Turing in front of me so can't remember the voltage range,
but when it's at either +2.5v or +5v then it'll be in the middle, when
it's +5v or +10v you should be locking hard right... make sense? The key is
turning the knob fully anti-clockwise first and positive voltage input.

Djandy82, 2019-02-21:
Thanks. It makes sende and i will try this out and hope this tip does the trix.

Djandy82, 2019-02-23:
Hi. I tested your tip a few times and still nothing. Could this be a chip failure?

Thonk-Steve, 2019-02-23:
If that is the only thing not working it's a lot more likely to be a
soldering issue, could you upload photos of both sides of both PCBs?