TomWhitwell/TuringMachine #111: Noise Generator breaks after patching anything

epijdemie, 2019-03-20:
So i finished a MK2 from Thong yesterday and was also stuck on the calibration bit like many other users. It did just not make any tone until the trimmer was far off to CCW side which was useless then because the noise generator did not produce any randomness. So i started to look at schematics and user postings and tried to nail this issue down to the Noise generator circuit itself:
looked at the NOISE out with an oscilloscope and it really did not produce any noise, but rather some slewed squarewave with some glitching.

So i changed T1 (3904) and viola i had a beautiful noise on the NOISE output!

But as soon as i patch anything either into trigger in or CV out the noise gets into this weird uniform shape again and the module does not behave as expected anymore.

So i change the 3904 again and the noise generator works again, so i assume something kills the 3904 or is this expected behavior?

I am at the 4th 3904 atm and the pads are getting tired, so i'd rather hear your input what might cause this.

Thonk-Steve, 2019-03-20:
Hi,

It sounds like maybe you have a short circuit somewhere else. Could you
upload photos of both sides of both PCBs? Or email to support@thonk.co.uk
Well lit, square on, and in focus photos please.

Cheers

Steve

epijdemie, 2019-03-20:
I think i might have isolated the problem.
One 4050 causes the 3904 to fry. It's probably my fault because i had one 4050 in his socket the wrong way around the first time (yeah, listen to the manual to sleep).

I just removed all IC's (because i might have damaged not only the one 4050 in question) and powered the board with a new 3904 and its producing beautiful noise again and i can just patch anything anywhere, the noise stays stable.

I will order a new set of IC's for the back board and report back once they arrived.

Soldering wise i would be surprised if there is a cold pad or a bridge, since this is not my first module and i am pretty confident at soldering. I probably suck at attention late at night tho :/

epijdemie, 2019-03-28:
Update: got new ICs from Mouser today and everything is looking good so far. Nice Noise output too and i can patch whatever i want. Noise stays, so a fried IC was probably causing the noise generator transistor to fail.

But now i am stuck at the same problem a lot of people have: Calibration. I just cannot get a lock before i almost completely eliminate the gain on the noise circuit and even if i am almost completely CCW on the trim pot, the sequence just does not get locked. it keeps playing random stuff or if too much CCW (almost to the end) nothing at all.

Any idea there?

epijdemie, 2019-03-28:
another update: during calibration my output does not sound as any of the other calibration videos suggest. it's always noise (like in the 12oclock position). never changes to a glitchy tone that needs to be fine tuned, just pure noise whereever i turn the big knob :(

i can try to obtain some more 3904 to test (so far i tested 3 batches with no success), bit it seems noise generation is not the issue maybe?

Thonk-Steve, 2019-03-29:
well perhaps, but it's still best we take a look at those photos first

epijdemie, 2019-03-29:
here are the pictures turing https://imgur.com/gallery/VHT3BrR

(i tested a lot of NPNs so i didnt risk to kill the pads and just solder the legs directly to old legremains)

epijdemie, 2019-04-01:
Ok, i read into the schematics and see the 4050 and 4081 are only vital for the expanders, so the TM should be perfectly working without them.

another thing i remembered from the first night build was that i plugged in headphones directly into the CV out (to listen to the output) and THIS in fact blew my transistor, not a faulty 4050 from putting it into the socket wrong. the original 4050 is actually still fine. It was the headphones that should have never gotten into the CV out. bad noob mistake when you are too tired already and wanna get things done in the lab in a hurry.

Now i really cannot figure out with the schematics why the transistor in the noise circuit broke when i patched headphones directly into CV out and what other components might be affected by this mistake. i already swapped all the ICs to no real success, so i guess it might be another component (a cap?) damaged that leads to this issue. in fact as several people have the problem of "no noise coming out" i suspect i was not the only one "listening" to the CV directly through headphones and not a proper output module.

well, i could now try to swap other parts of the circuit. any reference points of voltages i should have on certain nodes or terminals? or whats the noise level like at the TL072 thats at the 4016 which are both make the lock happen or not.

please help. i really would like this module to get to work properly :/

Thonk-Steve, 2019-04-01:
Hi, what were you expecting to hear on headphones? You know it only outputs
CV not audio?

epijdemie, 2019-04-01:
during the calibration process. you are supposed to "listen" to the CV output while you clock the module with audio rates. the "listening" cannot be too literal (i.e. headphones). must be with an output module. the calibration process (since you clock so fast that you actually hear glitchy sound) made me believe it would work with headphones directly as well, but obviously the resistance of phones are too much. yeah, a tired noob mistake. maybe you wanna warn others in your building document or calibration video that headphones directly are a no-go.

Thonk-Steve, 2019-04-01:
Ah apologies, I didn't realise you meant as part of the calibration. Could
you email on support@thonk.co.uk and give us some information on your
original order from us and we will work out a way to resolve this.

epijdemie, 2019-04-15:
Got a replacement from Thonk and it works flawlessly. Thanks a ton!