TomWhitwell/TuringMachine #114: Turing Machine stalls sometimes?

rubyglow, 2019-04-01:
Hi Tom/crew

Turing Machine is great obviously, and we're blessed to have **two** software implementations of it in VCV Rack. However they both share the same problem, and now someone confirmed that actually the hardware Turing has the same problem: Sometimes it simply stalls for a period, stops producing new notes, and then after a while it starts again. The someone said that it's usually alleviated by "twiddling the knobs" which kicks it back to life. Are you aware of this? The thread about it is here: https://community.vcvrack.com/t/the-turing-machine-why-does-it-stop-producing-notes/2647

Thanks a bunch, and thanks for Turing, which is still amazing!

deepiain, 2019-04-01:
It’s obviously trying to pass the Turing test... 🤣

(Hopefully someone gets this terrible joke)

rubyglow, 2019-04-01:
> It’s obviously trying to pass the Turing test.

I'm **sure** you mean the halting problem 😜

deepiain, 2019-04-01:
https://www.sciencealert.com/robots-could-crack-the-turing-test-just-by-staying-quiet

Thonk-Steve, 2019-04-01:
Hi Lars,

Just to be clear, do you mean this is happening when the module is 'locked'
or when the knob is somewhere in the middle?

Steve

TomWhitwell, 2019-04-01:
Hi Lars, 

I think - given that this happens both in the hardware (all logic chips, no code at all) and the software (all digital code) versions, that it may be psychological, not physical. 

The module is genuinely random (not even pseudorandom like software) - and because of that it's quite possible that it could go a second or two without 'flipping' a bit. Remember that only the states when the clock 'ticks' are important - it's like flipping a coin and getting heads ten times in a row. It's unlikely, but it can definitely happen, particularly if you're flipping the coin 8 times a second (at 120bpm).  

I've definitely experienced the 'move the pot' effect, but I'm fairly certain - given it happens on both versions - that this is simply a second or two when the sequence is 00000000 and no bits get flipped, then a bit is flipped after you've moved the pot. The small pot movement makes no difference. 

When we're focused and waiting for something to happen, time stretches out, 100ms seems like 1s - like the elevator / mirror scenario (https://medium.com/fluxx-studio-notes/five-things-elevators-teach-us-about-design-psychology-and-hats-60558c80b8ad) 

That's what I think, anyway. If it was only in the hardware I'd imagine it was some capacitance/inductance/CMOS quirk weirdness that I don't understand, but if it's also in the software that seems very unlikely. 

Tom

rubyglow, 2019-04-01:
> do you mean this is happening when the module is 'locked' or when the knob is somewhere in the middle?

I don't recall it happening when the sequence was completely locked, like the guy with the hardware also said. It seems to mostly happen when the knob is somewhere in the middle. But I haven't taken careful notes though.

rubyglow, 2019-04-01:
> it may be psychological, not physical.

Thanks for your reply Tom!

Heh... I didn't expect that reply and I definately see what you're saying, and I didn't actually realize that Turing has no code at all. That certainly strengthens your argument. And yes, statistically, given enough time, randomness will actually produce a sequence of nothing. Which is obviously a bit annoying because that's not great for music.

I haven't taken careful notes, but my most vivid memory of this would be something like the Turing being clocked at around 90 BPM maybe, and having occurences where it would stall for up to what seems like 10 seconds or so. But I definately see your psychological explanation, and that even when it gets going again, it seems like a while before something is happening.

Well, I think that probably the only thing for me to do is to close this one, especially seing that there is actually no embedded code that could be buggy. It's still a bit of a shame music making wise, on the occasions that it happens, but there are worse things in life, and the Turing is still very enjoyable :-)

So thanks very much for your reply, and for making the Turing open, so we can enjoy it in Rack as well. I certainly do!

Thonk-Steve, 2019-04-01:
It would be interesting to think how to patch around it though, easy ways
of detecting it and make sure something else happens elsewhere in the
synth? But then it doesn't happen often enough to warrant the effort. C'est
la vie!

rubyglow, 2019-04-01:
> how to patch around it

Yeah, probably difficult to detect and work around in the patch as such, but it does give me the idea, that in software it might be easy to detect this condition and to mitigate it by temporarily speeding up the generation, or something like that. On just that occasion the software might actually have an edge over the hardware. Hmm.... something to mull over with the developers if they're interested.

Thonk-Steve, 2019-04-01:
After 7 years with the Turing I just consider it part of the weather to be
honest, like the 'divebomb' effect that will occur at times too :)

TomWhitwell, 2019-04-02:
In the Buchla 266 (there is a little capacitor / resistor doohickey that pulls up one bit when the module starts up, so that it never starts at zero. I wonder if that would work here. You can see it in the top right corner of the schematic here http://www.synthfool.com/docs/Buchla/Buchla_200/266/Buchla_2660_1.jpg

closed 2019-04-02 by rubyglow

Skrylar, 2019-04-02:
> ps... for a long time I was convinced that Apple iPod shuffle wasn't properly random, that it was shuffling between a limited selection of albums to conserve battery life or something. I'm pretty sure I was wrong, and to make something that appears random, you sometimes have to make it less random.

Apple actually changed the random/shuffle algorithm at one point. It used to use a "fully random" mechanism but people found it too clumpy (as random numbers often are) and switched to a skewed randomizer.