TomWhitwell/TuringMachine #115: Sharp spikes on cv changes

kimitobo, 2019-04-13:
Hi there,

Freshly soldered Turing Machine spikes all cv changes very sharply, effectively clicking the pitch out on every cv change clock. Especially on ascending pitches. Prior to this I fixed the same "reversed-scale-pot" (issue #87) successfully relsoldering R26 (to Pin2 on TL074) after which the spiking/clickig appeared. Triple checked that area of circuit, any chance problem being elsewhere? Scope image attached, I have also video, but should be pretty obvious from the pic.

[image]

Thonk-Steve, 2019-04-13:
What are you clocking the Turing with? Does changing the clock source
change that at all?

kimitobo, 2019-04-13:
Tried few clocks, Echophon & Maths(EOC) and one external sequencer clock, pretty much same. After few different sequences, it's quite consistent that the peaks occure when the pitch is rising, decending cv does not have the peaks.

Thonk-Steve, 2019-04-13:
If you attenuate the input clock signal does it affect the amplitude of the
spike?

kimitobo, 2019-04-13:
Yes! That did it! Bit of modular noob still, did not occur to me you can attenuate clock, well of course you can :) So clock down to about 1v with Maths did it. I guess I can add voltage divider resistor pair directly on the module to save my my Maths channel? Anyway, thank you so much for quick responses. Quality customer service :)

Thonk-Steve, 2019-04-13:
Ah well that at least gives you a workaround... BUT it does suggest you
maybe have a.short circuit somewhere between the clock and the main output
as this shouldn't be happening and you shouldn't need to have to do this.

kimitobo, 2019-04-13:
Ok, I'll have a closer look following the schematic.

kimitobo, 2020-04-27:
Now year later had another look on the spike thing and realised that I have (I think) the random voltages doubled, so voltage change twice in every clock. Also noticed, the last four LEDs has low-brightness steady blink on the clock. Likely all related. The "second random" is louder and peaks through and also creates this not so random "ping-pong" with the first random.
https://photos.app.goo.gl/TkqWVvjpzXYSu4dB9

Could you point me to the part of schematic / board where should I have a look. Did few re-solders but more specific area tip would be great. Or am I seeing things, and all are as they should be :) Thanks!

TomWhitwell, 2020-04-27:
Maybe start at the clock input? 
[image]

kimitobo, 2020-04-28:
Problem solved. I was puzzled with the exact four last leds (or actually first I guess: B1-4) even clock blink and followed to the buffer and and chips. One corner leg was not quite in. I was so eager to pop it out, fix and try it that I did not pay attention if it was the buffer or and chip and which corner, so apologies for bad debugging report. Leaving a corner leg floating in the air creates this bug that is almost a feature: double random with higher pitch clock beat :) 

Anyway, thanks again for a swift response!

closed 2020-04-28 by kimitobo