TomWhitwell/TuringMachine #116: One led is not active

paulfreeman, 2019-04-22:


paulfreeman, 2019-04-22:
This is the third led from the top left, it doesn’t light.

No current is passing through when I test with a multimeter. I’ve checked the soldering including on the dac, but no joy. Any other ideas for debugging?

Thonk-Steve, 2019-04-22:
Do you have a diode tester on your multimeter? Sometimes they provide just
enough current to light an LED very slightly so you can tell at least if it
is working. If you do maybe power off the module and try it on one of the
LEDs that is working first. Also if you look carefully inside the LED you
can see the polarity from the shape of the element inside, make sure it
matches the other ones.

 If you think the LED got damaged during the build you can also buy a new
one here -
https://www.thonk.co.uk/shop/new-style-flat-top-leds-turing-pulses-vactrol-mix/

paulfreeman, 2019-04-22:
Thanks Steve,  yes I used  the diode tester and  maybe was reckless as I measured the current across them as the board was running. 

The problem diode  was not getting any data, no current across  its terminals.  The others were showing a voltage when they were switching on.  

I did manage to burn out the diode next to the one that wasn't come on. Testing that with the board off I get a voltage transmitted so its now no longer a diode and doesn't light, I'll buy a replacement for the one I've zapped.  I take it that its not a good idea to use the diode tester on the multimeter to test a live module!

On the schematic I see that these diodes are connected to the dac outputs via resistors and I've checked the connectivity through to the dac for the original problem diode and it looks OK. 

Could there  be  a problem with the dac?  The socket soldering seems sound.

paulfreeman, 2019-04-29:
Just to say, after replacing both diodes I tracked back on the schematic to the resistors connected to the diodes and found a 'dead' one which I replaced. Now all diodes are working.