TomWhitwell/TuringMachine #117: Issue Turing machine runs only when Write is pushed

moudular, 2019-04-25:
Hello

the Turing only works after I press Write
I check all the parts and it seems ok
I m ready to do it again
Where should I start ?
put a video to show what s going on, here ->
https://www.youtube.com/watch?v=xBMCDjvvuFQ

I started with mikrophonie a few weeks ago  then a Zlob LFO and then the turing machine

thanks for all

Thonk-Steve, 2019-04-30:
Hey, best thing to do first is to upload photos of both sides of both PCBs
or email them to support@thonk.co.uk

moudular, 2019-05-01:
I've sent you pictures
I hope we'll find the issue

moudular, 2019-05-09:
Hello,
Here's a link to differents photos of the 2 populated PCB
<https://photos.app.goo.gl/2UeUUS1uDSt3AAQNA>https://photos.app.goo.gl/2UeUUS1uDSt3AAQNA

Cordialement
PM Roy
07 86 32 09 45

De : Thonk-Steve
Envoyé : mardi 30 avril à 12:35
Objet : Re: [TomWhitwell/TuringMachine] Issue Turing machine runs only when Write is pushed (#117)
À : TomWhitwell/TuringMachine
Cc : moudular, Author

Hey, best thing to do first is to upload photos of both sides of both PCBs
or email them to support@thonk.co.uk

moudular, 2019-05-16:
Please I need help 
Thanks