TomWhitwell/TuringMachine #118: No sound output and only two leds

TheCelloFello, 2019-08-01:
[Video](https://youtu.be/8AF2zBRdqm4) 

[image]
[image]
[image]
[image]

When clocked by square wave from VCO, only the first two leds are lit and there is no sound. Reflowed all solder joints, checked transistors.

TheCelloFello, 2019-08-13:
Any ideas?

Thonk-Steve, 2019-08-14:
Hey,

These aren't Thonk boards so I can't vouch for the quality of them, so just
wondering which BOM you used to purchase from?

Regards

Steve

TheCelloFello, 2019-08-15:
Thanks for the reply, I used the BOM in this github. https://docs.google.com/spreadsheets/d/1FUUejrNd6qe-PUmlLi0BAgiGKWX9DxQH3Jt5qFOxPAQ/edit#gid=0

I ordered all parts from Tayda except for fuses (576-250R120, 576-60R010XU), ferrite beads (623-2743001112LF), and some ICs (595-CD4015BE, 595-CD4016BE, 926-DAC0800LCN/NOPB).

Again, thank you for your support.

TheCelloFello, 2019-08-15:
The board was made by Magpie Modular, I believe his work is reputable.

TomWhitwell, 2019-08-15:
I can’t see anything obvious in the soldering - what happened with the power header? From the video, it could be...
- an issue with the 6 unlit LEDs - orientation, something with their matching Resistors. 
- an issue - missing connection, burnt out chip, unseated chip with the 4015 shift register? 
- something else I can’t think of

TheCelloFello, 2019-08-15:
ahah just me being an idiot, I soldered the backpack connector header to the power header spot and had to remove it. All of the leds are the same orientation, reseated the 4015 and no change. Nothing appears to be burnt, which ICs would you recommend I replace first?

TomWhitwell, 2019-08-15:
You can swap the 4015s to see if either is faulty, but - assuming you didn’t somehow short-circuit or reverse power to the chips, fault chips are much much much less common than soldering / position / orientation mistakes. Have you reported your symptoms to Magpie for support there?

TheCelloFello, 2019-08-17:
Ok, swapped 4015s and had the same issue. Visually functioning but with only the first two leds, no CV output. Pulse and noise are both giving outputs. I will double check the orientation of all of the parts just to make sure. I will submit a report later today to Magpie. I used this build document: https://thonk.co.uk/documents/turing2016/Turing_Machine_Mkii_Build_Doc_v1.01a.pdf

TomWhitwell, 2019-08-17:
That's very odd - I've never seen symptoms like this.