TomWhitwell/TuringMachine #119: Weird behaviour - not locking, skipping LEDs, irregular speed

mcleinn, 2019-09-21:
Hello! I recently bought and assembled a Musical Thing Modular TM (PCB + panel kit), and am getting quite weird behaviour.  Checked and double checked resistors, replaced the DAC and swapped the other ICs. 

Symptoms: 

a) The TM seems to lock, but it usually doesn't lock what it is supposed to lock (when comparing with a 2HP TM), and after 10 seconds or so will lose the locked melody. 

b) In the calibration phase (audio-rate square wave input), when turing the knob, the output stabilized, but somewhat noisy, not with a clear sound (sometimes it lets out clear sound for a second). LEDs all light up, but not constant red, but slightly flickering (all 8 exactly the same). 

c) During usage (clock input), more often than not, the 3rd LED from the right is either OFF or showing exactly the value of the 4rd LED from the right, ie. 3rd and 4th switch on and go off always together! 

d) Often, but not always, the sound seems to "speed up" when the first three LEDs from the left light up, and then slows down.

You can see this on the following videos:
1. https://drive.google.com/open?id=12NKmWcMemF_es9J5ZLdoodUWUklNrZH-
2. https://drive.google.com/open?id=12WJqKo3S09yyAHj4-Dnpmr3oDinU4IB0
3. https://drive.google.com/open?id=129as-VXoKoVlo6pi4O3ygLsDtFPTCMAI
4. https://drive.google.com/open?id=129aYQmDjT4s1RjE_uMl-p_hSTzv7DL5q

Do you have idea where to lock for the problem? Especially the LED behaviour seems to be quirte suspicious. I can post more videos/photos if you tell me what to look for.
Thank you

Tobias

Thonk-Steve, 2019-09-21:
Hey,

To start with could you take good photos of both sides of both PCBs please?

Thanks

Steve
Thonk

mcleinn, 2019-09-21:
Here they are: https://drive.google.com/open?id=1YqjljttQ3fKO5RsBqAOyn03MDm-hw2sJ

TomWhitwell, 2019-09-22:
Hi Tobias, 
The videos definitely look odd - what are you using to clock the Turing? I've never seen that behaviour before. Strange clock behaviour can cause lots of odd symptoms - which possibly could explain the skipping steps, but it does seem odd.   

In the photos, I see lots of chip brands I've never seen before - a blank chip where there should be a TL074, a AM(?) brand DAC0800, some chips with Cyrillic  letters where the 4050 buffers should be. It's hard to say without knowing more about those chips, although the two 4015 chips see to be TI - they're definitely the most critical, so that's good. 

I'm afraid I have no idea about the 2hp module, that is a digital module, he just took the name without any approval, don't know what it does, and I've never seen one. The guy who made it had some rather confused ideas.

mcleinn, 2019-09-23:
I will try replacing all ICs with original ones. I'm based in a country where soviet analogues are the norm, and the engineer who helped me says they are "full analouges". However, I'll want to make sure this is really the case - the basic logic ICs should be fine, but not so sure about, for example, the DAC0800 (which seems to be from China). 

The clocking issue I will check again, maybe it was the Arturia Minibrute 2S which was playing tricks on me (it could be, that the "clock swing" function was on, which would kind of explain this). However, it wouldn't explain the locking issue - during calibration phase, I am using the audio square wave output anyway, not the clock.

Furthermore, the engineer who helped me, asked me to tell you the following: "If you put the scope to the pins 8, 7 or 6 of the DAC (signals B4,B3,B2) the output signal is changing, you can hear it. The inputs are digital, they come from digital sources - this shouldn't be the case." Does this tell you anything?

horstronic, 2019-11-15:
Hello Tobias, did you have any luck on solving your issue?
I'm having a problem with my touring machine that is somehow similar to yours, I just opened an issue (#123 ). I also have a problem with the same LED, maybe that's not a coincidence? 
Also I noticed the same thing with the output changing when I try to measure the Bits with my scope. By the way, I'm using an original DAC0800, so I guess yours is fine.

mcleinn, 2019-11-16:
Hello, not yet - I ordered the original ICs, but couldn't so far test them
out, for logistical reasons. Will try at beginning of next month. Maybe
your issue will shed some light on the problem. Regards

TomWhitwell, 2019-11-16:
Looking again at this - the description in the subject line 'not locking, skipping LEDs, irregular speed' does sound like a clock issue - I'd also check the whole clock input circuit - clocks with very jagged edges can cause inexplicable behaviour - imagine if the clock triggers 20 or 50 times on each transition, the module's output will seem very strange. 

The other thing that can case really strange global issues are power issues - either in the eurorack case, or coming into the board - check the diodes are all the right way round, the capacitors aren't confused with fuses etc. Your comments about scoping the DAC pins and output level made me wonder if there is some overall power problem.  
[image]

mcleinn, 2019-12-02:
I double checked the clock to make sure that no "swing" function is
enabled, and replaced the DAC0800 "analogue"
and several other ICs with the originals - the issues have disappeared. I
do have the curious issue, that when set
to 4 steps and locked, every ~16 steps a different note is played, while
the 3rd LED from the right goes on. The note
is repeated unchanged and reliably, as if it really was a longer sequence
which was being played. Any ideas on that?