TomWhitwell/TuringMachine #122: No Random out of CV out.

Vancooly, 2019-10-29:
Hello,

I'm having issues with the Turing Machine MK2 Kit

When i do the Calibration Process there seems to be no glitches in the output as shown in the video, the output is always a constant tone, and the Trim pot has no effect on the output.

The noise output does output white noise, and does react to the trimmer. With no noise on the trimmer being fully counter clockwise. And full noise output if the timmer is fully turned clockwise.

I have used multiple oscillators and clock sources, this doesn't change the output.

If the TM is fed a steady 120bpm clock and the Main Knob is at 12' there is no stream of random data. Only when i manually wiggle the big knob from 11 to 1 there is some change in the data. 

The Pulse out and CV in both seem to function normally .

I did a reflow several times and replaced al the IC's i could get my hands on.
( 4015, 4016, 4050, TL072).

I noticed some scratches on the PCB, can this be the issue here?

[image]
[image]
[image]
[image]

JeffGreenlee42, 2020-01-12:
I'm sorry no one has answered you yet.  I have the SAME issue.  I built my TM a few years ago and I have NEVER figured out the solution to this problem.   I replaced all the IC's  but no luck.. when I test the output voltages with a volt meter,  I only get slight differences in the random signals ... not the distinct voltages I would expect.    I keep going back to the Circuit diagram but I have never determined where the fault is. :(

Vancooly, 2020-01-13:
> I'm sorry no one has answered you yet. I have the SAME issue. I built my TM a few years ago and I have NEVER figured out the solution to this problem. I replaced all the IC's but no luck.. when I [...]

Thanks for your response.
I had contact with support from Thonk, and they helped me with testing it for shorts etc.
Unfortunately we couldn't find the problem and they asked me to send in the module, and it has been in the shop for the last 1,5 month due to the holiday season being very busy.
Hope hearing from them soon with an update, will update this issue with the findings.