TomWhitwell/TuringMachine #124: No/low CV output

gluethemoose, 2019-12-31:
Howdy

Just finished the build and everything seems to work except the output. I could complete the calibration although the level coming out was extremely low. I could only hear the signal by adding a ton of gain to my mixer. However, I could get the stable tone with the lock knob fully clockwise as well as an octave below with it fully counter-clockwise. 

When I run the TM, I get Pulse outs and Noise but the CV output is doesn't seem to put out anything other than very low level noise. See this image and note the voltage scale.

[image]

Here's the Pulse and Noise out respectively:
[image]
[image]

Any ideas where to start? Thanks in advance!

TomWhitwell, 2020-01-01:
Morning, and happy new year! 
In all these images you have the ‘scale’ knob on minimum which would give a very low output, but it might be something else...

gluethemoose, 2020-01-01:
Hi Tom

Happy New Year and thanks for your reply. I had another look moving through Scale’s entire range and the Output doesn’t change.

Sent from my phone, sorry for the typos, punctuation and brevity!

pjbulls, 2021-04-11:
Hiya, any follow-up on this? I have a similar issue where my CV output is very low in level, though everything works normally otherwise (I can lock in bits, attenuate the tiny output voltage that's there, etc.). I triple-checked all connections between the DAC, output buffer and output; everything is as expected. I also tried replacing both the DAC and the op-amp but no dice. The TL431 is providing 2.5V as expected and the resistances / connections for the DAC reference cicruit are also OK.

I have a perfectly working module sitting right next to it, the only difference between the two is that I measure 0V on the VREF+ pin of the working DAC and ~2V on the DAC of the faulty module. This has persisted after swapping out the chips though, maybe I have 2 dud DACs?

Thonk-Steve, 2021-04-11:
Also cross-posted here on Muffwiggler -
https://www.muffwiggler.com/forum/viewtopic.php?p=3506040#p3506040

Did you buy the parts from reputable sources? As per the warning in the
public BOM there seem to be dodgy DACs out there for sure:

https://docs.google.com/spreadsheets/d/1FUUejrNd6qe-PUmlLi0BAgiGKWX9DxQH3Jt5qFOxPAQ/edit?usp=sharing

pjbulls, 2021-04-12:
I'd like to think yes, but to be honest I've had them for so long that I don't remember. At least one from the same batch did work but that's not saying much.. I have one on the way straight from TI so if that fixes it we can be sure 👍

Thonk-Steve, 2021-04-12:
Cool, let us know what happens!

pjbulls, 2021-04-17:
> I'd like to think yes, but to be honest I've had them for so long that I don't remember. At least one from the same batch did work but that's not saying much.. I have one on the way straight from TI so if that fixes it we can be sure 👍

Follow-up on this: got some new DACs straight from TI & everything works as it should now (though I did replace the shunt as well for good measure). Guess some of my previous batch were indeed duds!