TomWhitwell/TuringMachine #126: Lock problem

mikisos, 2020-02-01:
Finished Turing Machine build. While trying to calibrate as stated in the video a problem occurs:

When the 1M trimmer is turned fully clockwise, the pattern locks on every position on the big knob. When trimmer is turned in any other position anti clockwise the pattern doesn’t lock at all, regardless of the big knob position. Gives a feeling that the trimmer scaling would be totally wrong or something.

Leds, noise and everything else seems to be working. Could you guide me towards a solution, please?

Pics attached. The extra jump wire from ferrite to elcap is because original lead was lost while desoldering an elcap which appeared to be too tall.

[image]
[image]
[image]
[image]
[image]
[image]

mikisos, 2020-02-04:
Hi again,
I’ve been going through the connections with the schematics. Should R22 connect to 1M trimmer? There is no connection so I may be having a cold joint there.

TomWhitwell, 2020-02-05:
From this schematic: https://musicthing.co.uk/collateral/TuringMachine_schematic.pdf there is no connection between R22 and the trimmer - but R32 should be connected. 

Tom

mikisos, 2020-02-06:
Hi,
sorry I was mistaken. What I was meaning to ask was that should there be a connection between R22 and 50K change pot because there is none.

I got another Turing to switch boards and thus be able to narrow the problem. My back pcb works well with another front board so the problem is definitely on my front panel.

Best,
Mikko

mikisos, 2020-02-06:
Hello again,
Sorry for the inconvenience. The connections seemed to be right after all. Just very difficult to calibrate -just a tad anti clockwise from the maximum trimmer position seemed to give the right result.

Thank you for help!
Best regards,
Mikko

closed 2020-02-07 by mikisos