TomWhitwell/TuringMachine #130: How critical is C15 value

paulfreeman, 2020-03-20:
C15 is meant to be 330pf but the kit item was only around 150pf,   I have a replacement from a ceramic capacitor kit that is around 300pf but I'm, wondering how important good tolerance is because this is in the clock input? 
 
My 300pf value is close to 10% tolerance. I am wondering if the 150pf one may have helped cause problems in the lack of randomness in my initial build.  I'm guessing not, but this is the only component that I've found that seems way out of whack.

Thonk-Steve, 2020-03-20:
Hi Paul,

Is this a kit from Thonk? Is it a V1 or V2 kit? Could you send a photo of
the part you received for C15 please?

Regards

Steve
Thonk

paulfreeman, 2020-03-20:
Hi, 

The part is from a Thonk Kit I bought last year and could never get to work.  I eventually buggered up the front panel trying different things, so I ordered a new front PCB and have been desoldering parts from the old one where they have values that are not easy to find from my supplies. 

I desoldered this last week so forgot the exact value I metered off the removed part.  Here are photos I’ve just taken of it.  

So it says 331 on it, which is 330 though I wouldn’t have known the units and now its metering 0.52nf  so, my bad its nearer 520pf, ie: I had my divisor/multiplier confused as its about 2x out of the desired value. 

It looks slightly different from all the other ceramics, more circular In form. 

It would be great if this might have contributed to my problem as I tried a lot of different things to sort it without results.  Hence I decided to go for a rebuild assuming I’d messed up a part somehow. 

Thanks! 

Paul

TomWhitwell, 2020-03-20:
Hi Paul, 
C15 is not critical at all - it's part of the clock input circuit that helps smooth noisy clocks, I'd be very surprised if that was a factor. I imagine that the module would work fine in 90% of cases without C15 installed.  
Sorry - I wish that was your answer
Tom

paulfreeman, 2020-03-20:
I suspected that was the case… clutching at straws. 

Most likely cause is an error in my construction somewhere along the line. 

Thanks for the quick response. 

Take care

Paul

TomWhitwell, 2020-03-20:
What symptoms are you seeing?

paulfreeman, 2020-03-20:
None currently as I’m rebuilding the front panel.  The symptoms were a lack of randomness. When tuning the module I found that even with trim pot adjustments the sequences eventually settled down to a simple alternating pattern. 
 
I hope to finish the rebuild over the next week and will see if the problem still exists. 

Paul

paulfreeman, 2020-03-30:
To my enormous pleasure the rebuild worked. Don't know what the original problem is doesn't matter now.  I did test all the components other than resistors and managed to re-use most semiconductors and caps.

TomWhitwell, 2020-03-30:
That's great news, congratulations!

closed 2020-03-30 by TomWhitwell