TomWhitwell/TuringMachine #133: Issues and questions on calibration procedure

Faye108, 2020-03-28:
I have a few questions about the calibration procedure. I did a lot of searching, but was unable to find any answers to these specific questions. I'm very grateful if someone could help me. 

**Question 1: **
When connecting the VCO to the TM clk the audio output at the TM sounds more like a buzzing at a much lower frequency than the audio output directly on the VCO (which is a beautiful square wave). Why is that? 

**Question 2:**
When I slowly turn the trimmer CCW until I get to a flickering free state of the LEDs, the LEDs are only dimly lit at the desired point. In the Thonk calibration video the LEDs are fully lit at that point (timemark 1:15-1:30). The only way to get the LEDs fully lit on my modue is by making a full AND quick CCW turn to the end of the trimmer, which is obviously more than necessary. Why is that? 

**Question 3:** 
When I think that I reached a stable position on the trimmer, I then start to wiggle around with the main knob. When I turn it somewhat CCW (a.e. up to midnight or 1 o'clock), and then slowly turn it back CW, between 3 and 5 o'clock the LEDs go off and the signal stops. In other words: between 3 and 5 o'clock is a dead area. This behavior only happens when I slowly turn the knob in CW direction. When I turn it quickly, the LEDs stay lit and the signal is still there. Why is that? 

**Question 4:** 
There is something else that I don't understand about the calibration procedure: The goal is an even signal without any glitches AND no flickering of the LEDs. 

Now in the first 6 seconds of the Thonk calibration video a calibrated TM is shown: the frequency on the VCO is demonstrated at different levels (the "coarse" knob on the Wiard Anti-Oscillator). It is clearly visible that there is heavy flickering on lower frequencies and zero flickering at higher frequencies on the TM LEDs.
This leaves me with the question, at what frequency I have to perform the calibration exactly.

On my research I found a reply which suggests that the frequency of the signal doesn't matter that much:
https://www.muffwiggler.com/forum/viewtopic.php?t=211795

This kinda stands in contradiction to the above observation.

Faye108, 2020-03-28:
Ok, this is getting interesting. Some updates:

- I did reflow every single joint on both PCBs. As it seems, both issues under question 2 + 3 are gone. Yay! 

- What remains is the issue under "Question 1": As mentioned above, the audio output is more like a noisy buzzing, than the actual square wave input from the VCO.
But I discovered that if I only touch the TM audio out jack with the tip of my stereo cable (instead of plugging it in), I can hear the square wave nice and clearly on the speakers, albeit at a very very low volume. 
Is there something that I'm completely missing here? Am I not supposed to plug in the speakers into the audio out jack labeled "Seq" for doing any testing?

TomWhitwell, 2020-03-28:
Q1: The frequency will be lower depending on the sequence length. You're getting a random wave (sequence) that repeats every 2/4/6/8 (sequence length) steps. If you change the length switch, you should hear the output frequency change. 

Overall, don't worry too much about the exact detail perfect calibration: You want to get it to the point where, at non-audio rates, sequences lock cleanly and unlock cleanly.

closed 2020-03-28 by TomWhitwell