TomWhitwell/TuringMachine #134: Turing Machine not calibrating, locks to pitch

skramsino, 2020-05-25:
Hi,

having tried various square waves to calibrate, I can´t stabilise the output.
When controlling pitch in an oscillator, randomisation "stops" at the highest pitch and circles around in quartertones. So, first theres´s a random sequence of about 4-6 notes, then it freezes at around the highest area.
Here are pictures of the pcb.

[image]
[image]
[image]
[image]

katspaugh, 2021-12-18:
Any luck fixing it, @skramsino?

skramsino, 2021-12-18:
> 

No.. gave up. Never found the mis-soldering. Will build a new one!