TomWhitwell/TuringMachine #137: Calibration issue - bad resistor?

cptnal, 2020-07-12:
When calibrating there's a very abrupt shift from noise to tone, to the extent it's impossible to dial in a range where it isn't either all random or all static. I've been through all the joints and, whilst not pretty, they look sound and not shorted. I've also checked as far as I can that all the components are in the right place.

However, R7 on the jack board shows 88K on the meter, and it should by 470K. The banding looks correct, so the question is is the resistor bad, and could that explain my calibration difficulties?

Thonk-Steve, 2020-07-12:
The best thing to do is post a photo of how R7 looks, as you can't reliably
measure resistance in circuit as it may be in parallel with another
resistor.

Steve

cptnal, 2020-07-12:
Hello, Steve. Thanks for responding so quickly. Here are some detail pics of the R7 area.

Cheers,
Al
[image]
[image]

iamdamosuzuki, 2020-07-18:
Not sure if this helps at all @cptnal, but I also have 88 Ohms across R7

cptnal, 2020-07-18:
Thanks, iamdamosuzuki (nice) but I don't think it does. What would be useful to know is whether a functional unit measures the same value across R7 when it's in circuit. Then we'd know whether that was normal or if it really is something to go on.

cptnal, 2020-07-21:
Update if anyone's still out there. Reflowed the pins on the main knob, which looked like they could do with more solder. Now I have no tone and no noise, only bleed from the clock signal.

TomWhitwell, 2020-07-21:
Do you have no noise from the noise output? What do you mean by ‘tone’?

cptnal, 2020-07-21:
Hi, Tom. Thanks for responding.

There's no sound from the noise output and I don't hear noise-transitioning-to-steady-tone at the main output when turning the trimmer. I was hearing these on my first attempt at calibrating, but after reflowing those joints they've disappeared.

Cheers,
Al

TomWhitwell, 2020-07-21:
R7 is part of the clock input so very unlikely to be a factor here. How frustrating. When you say ‘bleed from the clock’ - was that at audio rate?

cptnal, 2020-07-21:
Yes, I'm clocking it at audio rate per the calibration video. On reflection I think the bleed was from the muted channels on my mixer - there's nothing from the TM.

cptnal, 2020-07-22:
Update: Reseated a chip and now I have noise, which fades to silence as I turn the trimmer CCW. Still nothing at the CV output though. Also, the big knob appears to have no effect on anything.

TomWhitwell, 2020-07-22:
Do you get lights moving across the top LEDs when the clock is at sequencer rates and the main knob is at 12 o'clock?

cptnal, 2020-07-22:
Yes, I do. But the CV out doesn't appear to affect the pitch of the module it's plugged into, other than a slight wobble in time with the clock.

TomWhitwell, 2020-07-22:
OK, so the problem could be on the output: assuming you've turned the scale knob to 5 o'clock, check around there for any shorts or missing connections

cptnal, 2020-07-22:
All looks good down there I'm afraid. However, I've measured rather than listened and there's actually something coming out of the output. Around ten volts, and it appears to fluctuate +/- a tenth of a volt in time with the clock.

TomWhitwell, 2020-07-22:
If the LEDs are good but no CV output, then the problem is likely either the DAC0800 or the TL074 - check connections, look for shorts and chip orientation.

cptnal, 2020-07-22:
Looks OK, as far as I can tell...
[image]
[image]

cptnal, 2020-08-02:
Refusing to let this one go (the alternative is throwing £100 in the bin) so I tested continuity between the DAC0800 and TL074 legs, and their solder joints on the board - all beeping as they should.

I thought I'd test continuity between the leg of the output jack and the pins of those chips. Whether that tells us anything useful I don't know, but I thought I'd ask whether it's as expected. The yellow rings show continuity with the output.
[image]

cptnal, 2020-08-13:
Hello, folks. Is there any hope for this unit? Or is it a write-off?

TomWhitwell, 2020-08-13:
@cptnal those are GND points - were you testing the outside leg of the output jack - the one visible? If so, that's normal, because that's the ground pin. 
[image]

TomWhitwell, 2020-08-13:
So at the moment: 
Noise working? 
LEDs moving across when clocked at sequencer rates?
CV Output gives +10v, fluctuating with clock?
Does the output go up and down when you turn the little scale pot?

cptnal, 2020-08-13:
Yes to the first two questions.

The CV output follows the scale knob - 0V CCW, 1V CW, 5V at noon.

TomWhitwell, 2020-08-13:
OK, so the problem is probably in this space: Which is pretty much the area I pasted above. 

Pragmatically, I'd probably start by simply replacing the DAC0800 and the TL074 because that's quick and easy to do once you've sourced the parts. I don't know where in the world you are, if you're in the UK I'd normally recommend https://www.bitsbox.co.uk/ but they're on holiday.  

[image]

cptnal, 2020-08-13:
Many thanks, Tom. I'll spend some time on that at the weekend.

cptnal, 2020-08-29:
Hello, Tom. New chips sorted the output problem, thanks for that. That leaves the calibration issue, but I've discovered that by overriding the big knob with a CV input I can make the until behave normally. The joints on the three resistors involved, and the knob itself look OK. So I'm wondering whether it's a problem getting the +12V or -12V to the knob. Can you give me a clue how I might trouble shoot that? My first thought was to check the legs with a multimeter, but hard to do with the module assembled.

cptnal, 2020-09-01:
Update: Reflowed the joints on those resistors and the big pot anyway, and I've now been able to calibrate properly and the module's working perfectly. Thanks, all.

closed 2020-09-01 by cptnal