TomWhitwell/TuringMachine #138: Turing Machine mk I - random not working

MickGlossop, 2020-08-04:
Hi Tom - with the main knob in the centre position I'm not getting any randomisation. The TM behaves as if the knob is fully CCW (16 steps, or 32 with the switch UP), i.e. a constantly repeating series of CV voltages.

It's behaving as if the shift register is stuck. If I use the WRITE switch to clear the register, the CV output becomes constant as you would expect, but as soon as I release the WRITE switch, the CV output returns to the same sequence that was 'stuck' previously.

This behaviour is unaffected by un-plugging the backpack and Pluses/Voltages boards.

Where would you suggest I look - should I change a chip?

Cheers, Mick G.

MickGlossop, 2020-08-07:
I've replaced both the 4015 chips and that seems to have solved the problem.

Thonk-Steve, 2020-08-08:
Hi Mick,

Glad you solved it and sorry no one responded yet! Summer holidays aplenty
at Thonk and Music Thing.

Regards

Steve

MickGlossop, 2020-08-09:
No problem - we’re all entitled to some down time!

Cheers, Mick