TomWhitwell/TuringMachine #139: Turing Machine gets stuck on one note when locked.

saunlani, 2020-08-17:
Turing Machine works as expected, except nearly every time I turn the large knob to the right, it will only play a single note.  I don't believe this is expected.

I have checked the board for cold solder joints.

Any help is appreciated, thanks!

saunlani, 2020-08-19:
I've tried two brand new TL074CN's

TomWhitwell, 2020-08-19:
This might be an issue with one of the 4015 shift registers. Swap the two chips to see what happens. If the module becomes more broken, then replace those chips. If no change, the error may be elsewhere.

saunlani, 2020-08-19:
I also did that, no change.  I noticed an issue when calibrating though.  No sound is produced when the big knob is turned all the way to the right.

saunlani, 2020-08-19:
ended up reflowing everything on the front board and it works

TomWhitwell, 2020-08-20:
That’s great news

TomWhitwell, 2020-08-20:
That’s great news

closed 2020-08-20 by TomWhitwell