TomWhitwell/TuringMachine #141: Main Knob seems inoperative - no random seq and calibration issue

bluasango, 2020-09-27:
Hi!
I'm an absolute newbie to modular and diy (super combo). The turing machine is my 3rd diy project. The ones before worked flawlessly.
I think the main (big) knob is not working:

1) During calibration (acc. to the vimeo video) I get a constant frequency, but as I turn the main knob to either position, the tone stays stable.
2) When giving it a slow clock, and putting the OUT to 1V/OCT of my Mini Mod or my XR22 it creates a random sequence, but that stays locked all the time irrespective of the main knob position.

I'm using a clock from my DAW, I know one shouldn't, but I don't have a different one yet.
I am glad for any advice! 

[image]
[image]
[image]
[image]

bluasango, 2020-09-27:
https://vimeo.com/462294903
&
https://vimeo.com/462294894

TomWhitwell, 2020-09-27:
Are you getting noise from the noise output?

bluasango, 2020-09-27:
Hi Tom, thanks for that quick reply.

No! Didn't check that before.

bluasango, 2020-09-27:
no noise

TomWhitwell, 2020-09-27:
Ok that’s the problem - check the around the transistor 2n3904 and components around that, look for missing solder points or shorts around there, reflow anything that looks dodgy - also on the tl074 pins

bluasango, 2020-09-27:
Alright, will do so in the next hours and give an update then. Thanks!

bluasango, 2020-09-27:
I resoldered every single joint and replaced the 2n3904 with a spare one. Still the same problems. Still no noise out.
Can it have any other reason?

TomWhitwell, 2020-09-27:
Tweak the trimmer, because that is the gain for the noise generator - if it's turned all the way down you might not be able to hear it. 

This is the schematic for the Noise generator 

I know they're hard to read, but you can see it's a path from the transistor, through two gates on the TL074 op amp. That chip seems to be OK, because it's the same chip used in the clock input and the CV output, both of which appear to be working. 
 
[image]

bluasango, 2020-09-27:
Thank you. I'll look into that!

bluasango, 2020-09-27:
Yes. I moved the trimmer slightly right and there's noise right away! 
But it seems to not lock any sequence, it's only random. Also the sequence length is ignored now I think.

bluasango, 2020-09-27:
Sequence length is fine, sorry.

TomWhitwell, 2020-09-27:
Have you done this: https://vimeo.com/163160088

bluasango, 2020-09-28:
Yes several times. There is a mini threshold, where the sequence jumps from locked to random while tweaking the trimmer, irrespective of the main knob position. Could it be that simply the pot (main knob) is defective?

TomWhitwell, 2020-09-28:
Faulty pot is possible but not very likely. Do you have an offset in your system - something that can output +5 to -5 ish volts (like the middle channels on a Maths)? If so, you can put that into the CV input and see if the system works fine like that.

biospherix, 2021-06-26:
Really glad I came across this thread. I was having a very similar issue and noted on the schematic that the Electrolytic 10uF in C6 was on the noise path. I suspected this was the problem, changed it and now the unit (and Volts expansion) work perfectly. However, I couldn't find any small pot capacitors and had the settle with a tall pot bent over but it still works. Thanks. 

[image]