TomWhitwell/TuringMachine #142: Calibration Problem, No Sound at All

eminyu, 2020-10-22:
Hi,

I was super excited when I finish the Turing Machine, but whatever I do, I can not manage to calibrate it.
When I patch a square wave to clock input, nothing is happening, there is no sound, no light movement on the leds at all.

I checked the resistors (replaced 2 that looks suspicious), capasitors and solderings. everything looks fine. I changed the amp chips (TL074 and TL072) it's behaving the same.

Without calibration, I tried to use the module to see if everything is working or not. Clock input accepts clock, leds are working, all the outputs are working, but of course there is no magic is happening without the locking and randomisation. I tried to do calibrate without hearing the voice, but it seems an impossible task :)

Here are the pictures;
[image]
[image]
[image]
[image]

Can you please help me, what is the cause of the problem? Should I try to change the other IC's? I was super eager to have this wonderful module, but now I am feeling a bit frustrated. 

Thanks in advance, all the best.

TomWhitwell, 2020-10-23:
OK, some things to check: 

Are you getting noise from the noise output - and does that noise level change when you turn the trimmer? Your trimmer look a little damaged in the photo? 

Does the clock input cause the LED to light? 

When you say 'leds are working, all the outputs are working' - what do you mean? 

Tom

eminyu, 2020-10-23:
Yes, I get noise from the noise output. Noise level changes when I turn the trimmer. When the trimmer is on all the way counter-clockwise, there is no noise. Trimmer looks like a bit damaged because, I tried to calibrate it for a long time without hearing the sound :/ But it seems to be working, because it effects the noise level.

When I feed the clock input with a clock source (like Tap and Diviply outs from Start Up module), the clock input responses, leds are working according to the clock source. (Leds are responding with a clock source only)

Pulse output send pulses, scale knob works, it reduces and widens the voltage range. Output send pitch information etc.
All seems to be working.

Only problem is (weirdly), whatever I feed from the clock input, except a proper clock source, there is no response at all. No response to square waves, or any other audio source. There is no change on any of the leds. I can not hear any of them from the "Out" output. So it makes calibration impossible.

Sorry for my limited English, but I hope I managed to describe the problem.

Best,
Emin.

TomWhitwell, 2020-10-23:
Two thoughts: 
1. To calibrate with audio, you need to be in one of the longer sequence lengths - 8 or 16 
1. You can calibrate it as a sequencer without using an audio rate square wave - just ensure that the sequence is 'locked' at 5 o'clock and random at 12 o'clock

closed 2020-10-23 by TomWhitwell

eminyu, 2020-10-23:
Ok, thanks for the sequencer suggestion. 
After 2 hours trying to calibrate it, I think I found a sweet spot, it's kind of usable right now - but still there is a problem.

5 o'clock locks a sequence, 
7 o'clock locks a double sequence. 
12 o'clock produces random sequences, everything good so far but;
3 and 9 o'clock positions doesn't generate slightly random sequences. it continues to lock forever. even 1 and 11 is do not make a change, only 12 o'clock seems to differentiate the pattern.

I think, I am gonna leave it like this for a while and hope to fix it properly soon.
If you have any more thoughts and a suggestion, I would be more than grateful.

Thanks,
All the best.

TomWhitwell, 2020-10-23:
Sounds like you need a slightly louder noise level, but as you say, if it’s working, leave it.