TomWhitwell/TuringMachine #144: TM Stops working When Clock Is Connected

zburns12, 2020-10-27:
Hey Tom,

I’ve double and triple checked my solder joints and IC connections to the best of my ability.

When I plug in the module, everything lights up as it should. But when I put a clock into it, the Clock LED lights up and the rest of it shuts down. I’m using a square wave output from a VCO for clock. I can see the Scale knob working via oscilloscope before I plug in the clock. Turning off and back on resets the lights. A few times, I have seen the lights on top start running for a split second and then it all goes dark. I don’t get any more than a blip of sound and have tested at multiple trim pot positions. I’m at a loss of what to check now. 

Any ideas? 

Thanks

Video of behavior: https://vimeo.com/472416867 

[image]
[image]
[image]
[image]

TomWhitwell, 2020-10-27:
Hi Zach, sorry to hear that - could I ask where you got your boards from? 
Was the clock input in the video at audio rates, or like a sequencer pulse?

zburns12, 2020-10-27:
Hi Tom,

I got them from a US based store, https://www.sherpamodular.com/. 

The clock input was at audio rate, C4ish. I tried the square wave out of my Mangrove and TS-L.

TomWhitwell, 2020-10-27:
Hi, you'd have to ask Sherpa modular for support for these boards. I can't offer this kind of support on non-Thonk boards.

closed 2020-10-27 by TomWhitwell