TomWhitwell/TuringMachine #145: No randomnis and odd switch behaviour

pattex, 2020-11-17:
Hi there,

I just finished my Turing Machine, but sadly I did something wrong, I think.
I get nor real random sequence and also the Write switch is acting odd, I think. I made you a little video, because I'm not so sure how to describe the problems properly.
https://vimeo.com/480362268

I had a bit of trouble, ordering the right parts. So, this could be the problem.
Especially those two, I'm kinda unsure about:

C9: https://www.mouser.de/ProductDetail/vishay/d103k43y5ph6tj5r/?qs=wEmTtUuRSe6NCZwQxzdYuQ==&countrycode=DE&currencycode=EUR
I is very big and I remember from school, that there is a time component in capacitors. Could this cause the problem?

R5: https://www.mouser.de/ProductDetail/vishay/rcms0551001fhs14/?qs=PQpwOy0%2F8M2CZU9Kam2xVw==&countrycode=DE&currencycode=EUR
This should work, right?

Thank you very much. I didn't do something like that for a few years and I'm glad I did not burn the house to the ground.

Kind regards
Arne

Thonk-Steve, 2020-11-17:
Hi Arne,

I'll look in more detail later or let Tom respond, but those parts both
look fine to me, they'll work fine.

Steve

TomWhitwell, 2020-11-17:
Hi, sorry you're having problems. 

Switch behaviour looks normal to me. 

Did you try the calibration  from the video: https://vimeo.com/163160088 ? 

Where did you buy your PCBs?

pattex, 2020-11-17:
Wow, you guys are really fast (says the worlds worst open source maintainer)! Thank you both very much for your answers.

Good to hear that the switch behaviour is as expected. 

I tried to calibrate the unit like described in the video. But turning the trimmer doesn't seem to make any difference. So I turned it all the way clockwise for the video, because as I understand it, this generates "the most randomness", right?

I ordered the PCBs and all stuff I could from Thonk and the rest of the parts from Mouser.

Many thanks
Arne

TomWhitwell, 2020-11-17:
The trimmer level doesn't work quite like that. Check that the trimmer makes the noise output louder or quieter.

pattex, 2020-11-17:
> The trimmer level doesn't work quite like that. Check that the trimmer makes the noise output louder or quieter.

Yes, it does.

pattex, 2020-11-17:
In the schematic I found the note on the Ferrite Beads. I bought other beads, but in my understanding, this shouldn't be a problem. I bought those: https://www.mouser.de/ProductDetail/kemet/b-01-a/?qs=gt1LBUVyoHm3eH7lp3lW5Q==&countrycode=DE&currencycode=EUR

Also I checked, if using CV would work instead of the loop control knob, but it doesn't.

Thank you both very much. It is truly fantastic, what you are doing here.