TomWhitwell/TuringMachine #148: Volts only outputing very low Rang

TimiRose, 2021-01-23:
Hi everybody, dear Tom, I built two Volts modules. One seems to work like expected and it´s just great, the other one has a very limited range in the low range. Where could the fault be?
Greetings, Tim

TomWhitwell, 2021-01-23:
Hi, there’s not much to a Volts 

https://musicthing.co.uk/collateral/Volts_Schematic.pdf 

Check the op amp is installed correctly + seated correctly and look for shorts or missed pins 

Good luck!

TimiRose, 2021-01-25:
Hi Tom,
the only thing I can really think of is that I didn´t solder the resistors in the same directions. But otherwise everything seems fine as far as I can see...I attached two Fotos.
Greetings, Tim

[image]

[image]

TomWhitwell, 2021-01-25:
Not sure, but the solder points on the op amp look a little bit thin?

TimiRose, 2021-01-25:
Okay, I´ll redo those later. Difficult to get a good resolution photos with the phone. Thanks!

TimiRose, 2021-01-25:
Ok, sorry about that, the thin solder points on the op amp were the cause like you suspected. Works fine now.
Thanks for the quick help!!
Tim

closed 2021-01-25 by TomWhitwell