TomWhitwell/TuringMachine #149: Turing Machine faulty build

absenceofgravitas, 2021-02-14:
Hey guys,

Turing Machine is my second DIY build and unfortunately something has clearly gone wrong. It powers up but I'm not quite sure how to explain the issues as it doesn't appear to be doing _anything_ as expected. I made a [quick video ](https://youtu.be/gXkGR7Sr-fE) where I'm clocking it from a square wave as in the calibration video - the output is just going into my mixer.

I checked all the parts with a multimeter during the build but easily could have made a mistake. I've checked the  joints over and over again but can't see anything _obviously_ wrong aside from some sloppy soldering but nothing that looks obviously out of whack.

I bought all the components in the kit from Thonk, except for a CD401 chip that I bent whilst reseating all chips (issue was there before replacing this chip).

Would appreciate any help in troubleshooting!

[image]

[image]

[image]

[image]

TomWhitwell, 2021-02-14:
Hi, 
Sorry to hear that. Do you still have another mk2 Turing? If so, try swapping the front and back boards between the good and faulty modules - that should help identify which board the error is on. 

From the video it's hard to tell what the symptoms are - I'm not sure I'm seeing randomisation - let me know what's not working - also can be useful to test at normal clock rates just to get a sense of how it's working as a sequencer before starting calibration / fine tuning. 

Tom

absenceofgravitas, 2021-02-14:
Hey Tom,

thanks for your post - reading back on my original post I can see it was functionally useless in explaining my issues - I apologise.

 I've been playing around with the module now and have done as you suggested to see how it's working as a sequencer. I've another video [here](https://youtu.be/FqkNXDq9lMU) which I think shows that you're right - I can't seem to get any randomness? I just get this looping sequence no matter where the main knob is, and no matter how I adjust the trim?

absenceofgravitas, 2021-02-27:
Hey, apologies for the bump but would be very grateful if anyone else has any ideas for me to try!

TomWhitwell, 2021-02-27:
Yes, that doesn't seem to be working properly. 

There various threads here about noise and calibration. 

Check to see if you get noise from the noise output - and if the noise gets louder and quieter as you trim it. 

If not, that's the area to start looking at. 

Tom