TomWhitwell/TuringMachine #152: Working but impossible to calibrate and weird pulse and CV OUT (PCB & Osc pictures)

GrosBob, 2021-02-27:
Hi,

I'm facing two problems that occurred independently:

**1) right after building the module I couldn't calibrate it at all**. I did the following checks:
- Noise out OK
- Incoming pulse LFO is triggering the clock LED as expexted
- The noise trimmer seems to work as fully CW i can hear a kind of crushed noise that turns to sound as a white noise at 12 o'clock to end up fully CCW with nothing hearable, like a HP sweep from CW to CCW.
- Sequencer is running. I can add and remove bits using the switch. Pulse LED blinks on ON bits
- When the big knob is fully CW, the sequence is locked for hours. At 12, events start to be added to the sequence, that can be locked back the big knob is **very quickly** set back to fully CW, otherwise all bits are set to ON and CV to max.
- CV OUT is OK, at least its value change overtime
All this seem to be directly related to calibration issues if I understood well.
What is the frequency range of pulse wave I should feed as a clock to successfully calibrate the TM?

**2) today the TM started to behave strangely:**
- Pulse OUT triggers "normally" most of my modules (MI Plaits, DIY Befaco Rampage, DIY ADDAC T-networks), but I use TM to trigger all the 4 Proks (independently), a double trigger happens. The first on the ON bit (pulse LED On) and a second trigger occurs at the next bit, the next one only. I found out with a very basic oscilloscope that at the ON bit the trig signal was OK, but on the next bit there is a negative "trig" that manage to trigger the Proks. This negative trig only appears at n+1 bit (if n is the right trig. 
See the pict below: 
[image]

- Then I went back to control trig and CV (for Pitch) of Plaits and realise that CV out was also behaving differently. Instead of playing stable pitched notes, there was like a kind of portamento effect in between every ON events, but not a real portamento. Below two picts of the CV signal (one in DC mode, the other in AC). And here also there is a negative peak appearing after the ON bit. I suppose this neg peak is the cause of this protamento like effect.
[image]
[image]

I'm wondering if those two problems are linked. The appeared not at the same time. The double trig/portamento happen this morning when I powered the case.
Any ideas will be more than apreciated.

Here are the 4 pictures of the boards before someone ask me to post them.

Thanks

[image]
[image]
[image]
[image]

GrosBob, 2021-03-10:
Hello,

I finally manage to calibrate the locking of the TM. Every thing is working great for now.
Except the problem I mentionned concerning the  inverted'"gate signal coming out on the bit following a normal gate.
I didn't noticed it fist as the TM was triggering normally all the modules I have that can accept a gate signal.
I only notice this problem when triggering the Proks Drum modules (all the four). SO I'm wondering if the problem come from the TM itself or from the Proks that respond to a negative gate. 
Could you tell me if it is normal to have this negative trig at bite n+1?. And if this comes form a problem of the TM, where should I start to fix this.
Thanks in advance, I begin to feel desperate...

TomWhitwell, 2021-03-10:
Hi, 
I'm glad you have it working. 

I'm afraid I don't really understand what you mean by a negative gate. 

[image]

In this image you have the AC coupling on your oscilloscope

The negative pulse on your scope is simply the AC-coupled signal returning to zero. AC coupling shows CHANGE, not actual values. It's negative because it's going from n volts to zero volts, so is showing -n volts of CHANGE. 

On a DC couple scope it would simply fall from n to zero.  
 

You shouldn't get a double trigger - that would normally require the voltage to return to zero, then rise again above the threshold (likely 2-4 volts) 

Tom

jstenroos, 2021-03-10:
Hi,

What happens there is not a negative gate, but a very short ghost trigger on the n+1 position.

A picture tells it better:

It’s actually a design problem, to overcome that it would require a very slight delay on the AND operation to let the up and downward edges to settle. 
Most modules do not react to such short triggers, but some do, on my system MakeNoise Maths trigger input was the only one.

I solved the problem by adding small caps as low pass filters on the pulse outputs, also to the pulses extension module.

Best,
Jouni

GrosBob, 2021-03-10:
Thanks for your replies..
@Tom, I definitely agree and understand. The strange think is that only the Proks are triggered when the Pulse Led lit and once again on the next step even when it's off. This happens also when using the Pulse extension outs.
The explanation of Jouni makes sense to me. Could you tell me what cap value and exactly where you put them, especially on the Pulse extension. I would really appreciate.
Thanks in advance
Eric

GrosBob, 2021-03-10:
@ Jouni: You saved my life!!!
I just tried a 100nF directly from a patch cable and ...miracle!!
I successfully went down to 1nF. Time to solder...

Anyway, thank very much for your very constructive feedback.
All the best

Eric

closed 2021-03-10 by GrosBob