TomWhitwell/TuringMachine #155: No Random, reverts to same CV out

nickkjordan, 2021-03-20:
I'm getting a behavior similar to the video in #145. I've tried the calibration, but I'm getting only sporadic glitches and not any more or less adjusting the main knob. I am getting noise output with level proportional to the trimmer.

CV out can be modified with the switch, but it reverts to the same behavior which is an ascending then descending line, with pulses in the middle. I'm not really sure how to debug from here, but I've resoldered everything on the control board. I can attach a video of the behavior if it seems necessary.

nickkjordan, 2021-03-21:
Hopefully this helps someone out with similar issues and points them in the right direction, but I'm feeling a bit of :shame:

Because it seemed like everything was working besides the loop control knob (random vs locked), I took a look at the schematics and found it pretty much feeds directly into the TL072. I was going to switch it out, assuming I'd accidentally fried it, but realized it wasn't even there. Placed it, and it's working perfectly now.

closed 2021-03-21 by nickkjordan

TomWhitwell, 2021-03-21:
Mate, we’ve all done it

vfilby, 2021-04-22:
Hi Nick, 

Just wanted to let you know that this was very helpful to me!  Had exactly the same behavior and while my IC was in the socket I didn't fully solder in the pins in the socket.  I had just tacked the two corners of all the sockets then did the rest of the pins... and missed the last socket.  Once I actually soldered the pins everything works perfectly.  

Thanks!