TomWhitwell/TuringMachine #157: Problems with clock

nlj777, 2021-04-04:
Help! I’ve just finished the Turing machine build and cannot get clock to work when plugged in. All LEDS at the top light up but when I patch in a square wave the clock LED doesn’t light and no sound comes out. If I patch the output to a CV in on my VCO I hear a sound and the scale knob is working but if I then patch a gate signal into clock (from pams) and start pam there is no effect on the Turing machine. The clock led doesn’t light up. 

I’ve checked connections particuLarly round TL074 and can’t see any dodgy joints. Any ideas? 

[image]

[image]

TomWhitwell, 2021-04-04:
Sounds like a problem with the clock input, which is the bottom right on the top sheet of the schematic 
https://musicthing.co.uk/collateral/TuringMachine_schematic.pdf
Check all the components and points and that the tl074 is properly seated and the right way around.

nlj777, 2021-04-04:
Thanks for the quick response. I’ve checked all of those components and everything is in the right place and the right way round. I can’t see any problems with the solder joints so not sure why this isn’t working.

TomWhitwell, 2021-04-05:
If you have a multimeter you can check connectivity / trace the circuit - focus on that clock circuit

nlj777, 2021-04-05:
Hi - we have checked all the clock circuit with the multi meter and no problems. However the clock LED is not lighting when patched in

Chip TR074 is getting hot after 30 seconds to a minute, and the polarity on pin 5 and 6 is reversed when we first switch it on. There is no output from pin 7

Does this mean we have a faulty chip? 

Thanks

thismatters, 2025-01-25:
You've got at least 2 cold solder joints that I can see, toward the bottom of the first photograph. Pads connected to the ground plane require a great deal more heat to take solder. If you can't turn the temperature up on your soldering Iron consider getting some flux to help the solder wet.