TomWhitwell/TuringMachine #162: receiving clock but LEDs are remaining ON perminently and the output is not changing

ashreid23, 2021-07-12:
Just finished the TM ii and having some trouble getting it to produce any variance in voltage.

It will receive a clock, produce noise that can be attenuated.
When I turn it on (or try too clear with 'write') the sequence is immediately populated. The CV output can be changed with the scale trimmer pot but is otherwise constant.

I got it set up so I could probe with my oscilloscope while it was running. The two 4015 are receiving the clock but the outputs QA-D, you can see a slight variance in time with the clock pulse however they are remaining high.

FFER1 has a short to ground on Pin 14. I Have tried looking at the Eagle File, but I grew up on Kicad and cannot figure out if this is a fault or it should in fact be grounded? Is that essentially grounded at pulse out front when it is not connected. (I'm really not sure if I'm reading that correctly.)

video showing in general how it is operating:
I have it powered by the Joanalogue Test 3 to the left and the number displayed is the Ma draw.
https://youtu.be/nDoCevhVERQ

trying to probe the IC's:
https://youtu.be/lZHgFgAgTl8

Thonk-Steve, 2021-07-12:
Hey,

Can you confirm first where you bought the PCBs from, or was it a full kit?

Steve
Thonk

ashreid23, 2021-07-12:
Full kit from you guys

ashreid23, 2021-07-12:
Just managed to capture the CV output (and clock) with occasional randomness when powering it up / clearing with 'write' [image]
from what I can see it always seems to be the same 'descending' pattern.

Thonk-Steve, 2021-07-12:
Ok cool, if you start by sending photos of both pcbs on both sides we can
rule out any simple stuff. I'll leave Tom to comment on the IC probing 👍

ashreid23, 2021-07-12:
[image]

ashreid23, 2021-07-13:
Playing around this evening, I can create 'randomness'  by tapping SHIFTREG_MAIN pin 7 to gnd. However, I cannot get this to 'lock'. This seems obvious but just trying to rule things out.
Pin 2 and 10 on the 4016 remain high. However, I am still trying to understand what exactly that part of the circuit is doing and, well, how randomness is generated for the shift-register (start_of_loop), so I don't really know what to do with that information.

TomWhitwell, 2021-07-19:
Hi 

So the way that randomness works is that IC7B is a comparator - it compares the voltage from the main knob to the noise signal. If it's higher (or lower, can't remember), the signal from END_OF_LOOP is routed through SWITCH_ICA to to START_OF_LOOP = the bits in the shift register loop endlessly  
If it's lower (or higher) the signal from END_OF_LOOP is inverted - via the other gates of SWITCH_IC - and sent to START_OF_LOOP 

So:
all high = constantly looping 
all low = constantly inverting - double length loop 
randomly high and low  = sometimes inverts, sometimes not = randomisation 

I'm not sure if that helps with your specific problem - It does sound like it could be an issue around that 4016 switch, like you're never getting a connection between END_OF_LOOP and START_OF_LOOP - that would mean you could write a sequence but it would never repeat, and might mean randomisation didn't work. 

(BU)FFER1 pin 14 should be connected to ground, that's fine 

Did you say that even when you write bits with the switch, you don't get any change in the output voltage (assuming scale is turned up?) - that's odd 

In terms of troubleshooting, your board looks very clean and nicely soldered. Looks like there's a scratch near the back of the trimmer, make sure nothing is shorted there. I'd consider re-seating and re-checking all the ICs in their sockets, sometimes that can be an issue

ashreid23, 2021-07-22:
appreciate the response Tom,

I had posted about my problems over on [modwiggler](https://www.modwiggler.com/forum/viewtopic.php?f=17&t=250943) and with help debugging, In summary, I landed at there being a problem with the low pass filter pre START_OF_LOOP (R16/C4).

That 10k resistor, when measured (in-place on a fully populated board), was only good for 5k. I removed it and measured 10k between the two holes. (which rings true with the 5k with it in place). 
I tried removing SWITCH_IC and it all measures as you would expect, so am waiting for a replacement and hoping that is the fix.