TomWhitwell/TuringMachine #164: TM occasionally doesn't output any bits

TempleMind, 2021-10-03:
I have recently built a Turing Machine full kit and everything works fine, all functions are working as expected. However, it will occasionally not output any bits. This could happen with the main knob in any position, so either locked in either direction, in slip mode or full randomness. After about 10 seconds a new bit does get written (and when it's not outputting any bits I can use the write function to add one in). Once a new bit is written (either manually by me or by the TM) it kicks back in to producing patterns of bits as expected.

I have reflowed every solder joint, I can't see any issues with the board and have tested all IC pins with a multimeter to confirm they are making connections to the underside of the board through the IC sockets.

Could this be an IC issue, and if so which one might be the likely culprit? As mentioned, everything else works perfectly fine apart from this occasional drop in bits. Or is expected behaviour?

Thanks!

TomWhitwell, 2021-10-03:
Have you calibrated it? If the main knob is fully clockwise and the sequences changes, it may need to be calibrated. If the knob is at 12 o'clock and you get periods of 00000000 that is normal

TempleMind, 2021-10-03:
Hi Tom, thanks a lot for the quick reply. I have calibrated it and have just checked it again and all seems fine, so I'm getting a stable tone during calibration when the knob is fully clockwise and noise at 12 o'clock. When the knob is fully clockwise it locks, and I'm getting randomness at the 12 o'clock position. So sequences of 00000000 is as expected, just wanted to double check!

Thanks for your help!

closed 2021-10-03 by TempleMind