TomWhitwell/TuringMachine #165: Turing Machine Mk2 no pitch CV out, single voltage level

tim-gregory-pt, 2021-11-01:
I assembled a full kit bought from Thonk. I also have Volts and Pulses expanders (pre-built).
The TM mostly seems to work - I get flashing lights from the TM and the expanders, I get noise out at a fairly high volume, and I can use the trim-pot to get it to lock/latch using the calibration procedure video (but it needs to be 90-95% CCW to stabilise on the latching). Expanders work fine, and I'm getting some different notes out of the Volts.

Problem is that I don't get any variation in the tones output from the TM itself, it's always just a single tone.
The pitch is influenced by the scale trimmer, and I can lower or raise the pitch, but there is no variability.

Things I tried to fix it - reflowed everything twice, switched the 50k pot, switched the DAC, switched the 072 chip, switched the 4015's.

Any ideas where I could trouble-shoot? I feel like it's so close to working, I just need to work out what is preventing it from sending out a stream of different pitches instead of a single steady voltage.

TomWhitwell, 2021-11-01:
Hi Tim, sorry to hear that. 
Just checking - when you say tone, you mean pitch when the voltage from the CV output is connected to an oscillator? (Had to check) 
So if you have happy flashing LEDs but no CV output, the problem is somewhere here: 
[image]
It's the TL074 that powers the output voltage, so any problem / missing joint there, or if the TL431 voltage reference had been mistaken with another transistor-shaped component that could be a problem. Sometimes it's as simple as a missing solder joint on the output socket, or a short to ground somewhere around there.

tim-gregory-pt, 2021-11-01:
Hi Tom, thanks for responding here, and thanks for the detail on where to look. Yes, I am not getting pitch CV out, only a single pitch, which responds to the scale pot. I'll take a look more closely in the morning. If you're interested, here is a short video demonstrating what I'm getting https://youtu.be/qJ020EzP2i8

TomWhitwell, 2021-11-01:
Yes the video reinforces what I suggested above. Although a voltage is getting to the pot and out to the socket so the issue in the TL074 or DAC or supporting components

tim-gregory-pt, 2021-11-02:
I focussed on the area indicated and checked the TL431, tried another DAC, reflowed everything, checked all the resister values and looked for shorts but I'm still getting the same symptoms. Except now I seem to have lost 1 & 7 on my pulses!
I don't have any spare TL074 at the moment so I've ordered a few and I'll switch that out in a few days and see if it fixes the issue.

tim-gregory-pt, 2021-11-02:
Success! - I scrounged for a spare TL074 in my unbuilt kits and switched only this, and now everything is working ok, including the pulses lights that I lost.

closed 2021-11-02 by tim-gregory-pt