TomWhitwell/TuringMachine #166: Workaround for lack of clock reset functionality?

othmar52, 2021-12-08:
Recently I bought and built TM MKII and pulses DIY kits and i like it so much!
@TomWhitWell Thank you for those great modules!!!  
I am going to get the volts expander kit soon.

There is only one tiny thing that's a bit annoying and i am wondering how to "solve" this.
In my personal use i often make music with a bunch of other people with all kinds of synchronisable music equipment. we are all synced via midi-clock/cv-clock.
every now and then its necessary to restart the master clock in case one of the dozens involved devices has lost a clock tick.

As the Turing Machine has no "clock reset" (start locked sequence from step 1) a clock restart results in a time shifted loop which is very disappointing for all involved musicians.
I have to admit that this is a very exotic edge case but i am trying to find a solution/workaround for this.

The first thought i had was to create an extension module with a single pushbutton. pressing this button generates a single clock pulse which gets added to the existing clock input. Theoretically it would be possible to feed manual clock ticks until everything is in sync again.
Pretty sure this circuit is no rocket science but unfortunately my knowhow about electronics is not that good :/

Can i take the 5 Volts from the eurorack power bus and feed it directly (parallel to the main clock) into the clock input via a pushbutton?
Or will i destroy the TM with that approach?
Do i need a resistor?
Would this also be possible with the +12V potential of the power bus (to support other power supplies without 5V)?
Or is this not the right place for such a question?  

Or do you have any other idea how to get a locked sequence in sync again after a master clock start/stop?

Thonk-Steve, 2021-12-08:
The short answer is you can't reset the 'sequence' in the Turing Machine as
it doesn't remotely resemble a sequencer under the hood.

othmar52, 2021-12-12:
@Thonk-Steve thanks for your answer!  
In the meantime i have created an expander module for the TuringMachine:  
https://github.com/othmar52/eurorack-gatenudger  
currently only on the breadboard but everything is working like expected so now i am able to move the locked "sequence" left and right by adding/supressing single gate pulses via 2 pushbuttons...

closed 2021-12-12 by othmar52

TomWhitwell, 2021-12-12:
Amazing @othmar52 - keep in touch with progress

othmar52, 2022-02-06:
> Amazing @othmar52 - keep in touch with progress

@TomWhitwell I already have created a few eurorack modules, but this is the first time it feels like a **real** module  
With very little knowledge of electronics & stuff but with a lot of patience and passion i am proud to present [homogate](https://github.com/othmar52/eurorack-homogate)  
It's finished and everything works exactly like it was intended to be.  
Maybe i am going to create a short video which demonstrates the "nudge left/right" for TM's locked sequence soon...