TomWhitwell/TuringMachine #167: Turing Machine problem/strange behaviour

Heptalion, 2021-12-21:
Hi everyone !

I'm looking for help around my first DIY built, the Turing Machine Thonk full kit.

The build process was quite smooth, no apparent problem with the soldering. The only thing that makes me worried was in the instruction, 35th step, about the 100n capacitor, it says "you may have one 100, cap spare", but i don't, so maybe they just gave the righ amount in this kit. Anyways.

After cleaning the board, i plugged it, and tried to follow the "trimming the lock" video for calibrating the main knob.
That's where the strange behaviour kicks :
By plugging a square oscillator in the clock, and listening the out, I hear the noisy tune, but when I slowly turn the trimming CCW, the tune slowly disapear, leaving no sound and no LED. I noticed that even in full CW, not all the LED are lit.
I decided to use a slower LFO in the clock, even with 8 step selected, i only have 3 or 4 LED lit
Noise OUT sounds fine.

Do you have an idea of what is going on with this build?

I disassembled it, testing the diodes and resistor, they all looks fine, except the R7 showing 93k (instead of 470k). Do you think it is the problem?
I'll try to upload a video of the trimming process, and shots of the 2 PCB.

Thanks for your help ! don't hesitate to point out "basic" and common issue, i'm a noob in DIY. And it really frustrate me....

Heptalion, 2021-12-22:
Here is a video of the calibration process.
https://vimeo.com/manage/videos/659248416

We can notice that when I power up the system, all the LED are lit. It last until a plug a clock.
I use the nano ona in VCO mode, then I switch to LFO to see with more precision.
Noise output sounds fine, trimmer works.

Here are some shots :
https://ibb.co/6N1KMKq
https://ibb.co/MMpj1ww
https://ibb.co/580XK0v
https://ibb.co/kG93TkW

Heptalion, 2022-01-03:
I changed all the ICs (except the DAC) with some spare parts, the result is the same as the one shown in the video.
Some precision about the LED behaviour ;
When powered on with no clock, all the led are lit, and bright.
With some manual clock pulse, i can see this :
While introducing a null bit, and shifting it, i can see the brightness of the LED 4 and 5 dimming. When my null bit pass them, and switch off the LED 6 7 or 8, those three never lights up again

Heptalion, 2022-01-07:
Updated shot with a DSLR :)
[](https://ibb.co/qkwk546)
[](https://ibb.co/wZt0fSf)

thismatters, 2025-01-25:
It looks to me like there are some solder joints which could use some more solder. You want each pad to have enough solder to form a little cone up onto the leg.