TomWhitwell/TuringMachine #168: Calibration Issue: Trim pot doesn't lock until fully CCW no variable random output

Guanavia, 2022-01-23:
Hello, 

I've just finished the build for the Turing Machine MkII and have a calibration issue I haven't found a good solution to in the forum yet (If I missed it please let me know). 

**Problem**
The build went overall very well. I did realize that I accidentally swapped the main knob pot and the scale pot. Desoldering didn't work as well as I'd hoped and per the BOM they're both linear 50k pots so I expect they'd work while swapped until I can get replacements. I followed the calibration procedure video and with a square wave into the clock and the trim pot fully CW I get a random tonal pattern. As I turn the trim pot I don't get a locked tone until the pot is nearly all the way CCW. 

Once I do get that tone and patch with a clock at 120bpm the LEDs make it look like I should be getting a random pattern but that pattern barely changes as I sweep the main knob. There also isn't much change in the randomness of the LEDs and when patching it through a VCO it will play 3 to 5 notes, pause for a few seconds, then play the same notes again.

**Troubleshooting Steps**
The noise output does produce noise as expected. The Scale knob seems to sweep through a pitch curve. I've checked all the solder joints and they all appear to be solid and without issue. As I mentioned the main and scale pots are swapped but based on my reading of the BOM they're both linear and 50k so it looks like they're interchangeable.

Are those pots not interchangeable for some reason? If that shouldn't be the issue what else can I troubleshoot?

Thanks for all the help in advance, really excited this up and running!

Thonk-Steve, 2022-01-24:
Hi,

Yes pots swapped should function ok. What signal are you using for the
clock in to calibrate with?

When you say it does calibrate properly how does it actually function when
clocked at say 120bpm and used to generate a looping sequence?

Regards

Steve

Guanavia, 2022-01-24:
Hi Steve. Thanks for the reply. 

I'm using a square wave from a WMD Spectrum (pulswave out with pulse width set at 12 o'clock) to clock the calibration. 

I'm sorry if I mistyped that, I mean that it 'doesn't' calibrate properly. The calibration video specifies that turning the trim pot too far CCW will result in very little variation in the randomness. This seems to be the primary issue. The trim pot is almost all the way CCW and I am definitely getting very little randomness when turning the main pot, even when it should be completely random at 12 o'clock. And when turning the scale knob all that happens is the pitch of the pattern it's playing decreases. A more detailed description of behavior when clocked at 120bpm is as follows:

**Operation after following calibration instructions**
-TRIM POT: In order to calibrate to a consistent ton the trim pot has been turned almost entirely CCW. There is a minute amount of distance to full CCW, less than a degree of rotation. After having attempted to dial in the calibration multiple times that distance remains roughly the same.
-CLOCK: Clocked with a trigger out from a TR8S at (I believe) 24ppqn as well as with that same trigger signal run through a 4MS clock divider at different divisions. The tempo changes as expected and the change in tempo does not affect the randomness.
-SIGNAL CHAIN:  Main out run into the 1V/oct of the WMD Spectrum as well as a Plaits set to a standard waveform. The signal chains function as expected.
-SCALE KNOB: Calibrated with Scale knob all the way CW. When turning the scale knob on a constant tone it sweeps through a frequency spectrum as would be expected. When turning the knob with a 120bpm clock it changes the pitch of the pattern. So that appears to be working normally. 
-LENGTH KNOB: Have left this knob at 8 after calibration.
-MAIN KNOB: When the knob is at 12 o'clock and should be full random it will play a sequencer of notes in ascending order from low to high. It starts on the same note, end ends on the same note usually repeating the last note multiple times. It feels like it is probably an 8 note sequence and the LEDs appear to be showing an 8 note sequence. If I turn the main knob all the way CW to lock the sequence it will lock on the last note of the sequence and just keep repeating that note. 

**Attempting to change trim pot to 'non-calibrated' state**
-Given that the calibration directions do say that having the trim pot too far CCW will lead to lack of randomness, and lack of randomness appears to be my biggest issue, I test the unit with the trim pot in the center of it's range. 
-When calibrating with a square wave it is definitely not outputting a constant tone but instead just what sounds like a noisy bit-stream.
-When patching with a standard 120bpm tempo I get nearly identical bahavior as described when the trim pot is calibrated fully CCW and producing a contant square wave tone.

Guanavia, 2022-01-24:
Here's a short video of the behavior.

https://user-images.githubusercontent.com/98287874/150810032-a3bbce53-52d3-4644-b632-d5d2824a7a18.mp4

Guanavia, 2022-01-28:
@Thonk-Steve Hi Steve. Based on that information any idea what could be wrong or how I could start working towards a solution?  I'm quite excited to get to know the turing machine and would love to figure this out. Thanks!

fxwiegand, 2022-03-14:
I have the same behavior getting almost no randomization with the trim pot fully ccw and no proper locking of sequences when the trim pot is in any other position.

supermartincho, 2022-08-15:
hey guys i have the same problem. Did you resolve it?

Guanavia, 2022-08-15:
I have not.

I've tried recalibrating a couple times and was eventually able to get it
where it was "mostly" random, and would pseudo lock for a few phrases. I
decided that maybe I screwed something up in the build and decided to let
it be for awhile and use the "evolving randomness" when it made sense in
whatever I was creating.

I may just eventually replace the unit with one from reverb that's tested
and working, but while I love the simplicity and interface of Turing I also
have a Marbles which is working well for now.