TomWhitwell/TuringMachine #170: Unusual effect on powering up

davb2022, 2022-03-05:
I have just built a Turing Machine and have obliviously taken a wrong turn somewhere.  When I turn on the power supply my Tiptop Zeus and any of my other modules with led's flash on and off at a steady rate.  I didn't leave it on for too long as I didn't want to blow anything, has anybody any ideas?

Thonk-Steve, 2022-03-05:
Hey, it sounds like you may have a short circuit somewhere in the PCB,
although it's not normally so intense! Do you have a digital multimeter?
Where did you buy the PCB?

davb2022, 2022-03-05:
Sorted Hurrah! Unfortunately, in the excitement of almost being finished I had mounted the ic socket for DAC0800 the wrong way round and so I plugged the ic in the wrong way round, DOH.

davb2022, 2022-03-05:
Thanks Steve for taking the time to answer, I actually bought the kit from your business.  Must try harder next time.

closed 2022-03-05 by davb2022

Thonk-Steve, 2022-03-05:
Oh don't worry we've all done that plenty of times :) glad it's working!