TomWhitwell/TuringMachine #172: Turing Machine MK2, no signal from output

Kazanthul, 2022-03-09:
Hello Tom,

I just bought myself a prefabricated Turing Machine MK2 on Reverb and unfortunately no signal is coming from the main output. All the other ins/outs are functionnal but sadly the best one is not... Is there any way to check what's going on with the module knowing that I'm absolutely not knowledgeable about electronics whatsoever... I contacted the seller but no answer so far.

Is that a common issue ?

Thank you in advance for your help,

All the best,

Jim

TomWhitwell, 2022-03-09:
Hi Jim, you should get a voltage output - not audio - from the main out, which is attenuated by the small scale pot - so ensure that is turned up. Let me know if you're still having trouble - although it might be best to contact the Reverb seller.

Kazanthul, 2022-03-09:
Thanks for the quick reply !

I just checked it, still nothing unfortunately. I've used led cable to double check and obviously there's no voltage at all...