TomWhitwell/TuringMachine #174: Turing machine - Cutting out and trimpot calibration issues

tspear, 2022-03-14:
Hi, I finished my TM build recently and have a few issues which I'd like some advice with.

1. Following the calibration instructions on Vimeo with audio rate clocking from a pulse output, the audio output only becomes stable when the trimmer is almost completely CCW.
2. Starting at fully CW, The trimmer pot sticks when reaching about 2/3 CCW which is roughly where the 'sweet spot' is and needs extra force to get past the sticking point.
3. When the trimmer is around that point, the module seems to cut out completely for a short time, especially when the main knob is between 12 o'clock and 5 oclock. I've attached a photo of a possible dodgy joint on one of the ferrite beads. Wondering if this is the cause?

I've made a couple of videos to show the issues:
[Turing 1 - Audio rate clock with Pulse wave](https://youtu.be/HxhBqd1tgww) - point 3 happens around 1:47
[Turing 1 - normal clock ](https://youtu.be/CJZqvbowJZQ)

Also attached photo's of the boards.

thanks for any advice.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

TomWhitwell, 2022-03-20:
“ The trimmer pot sticks when reaching about 2/3 CCW which is roughly where the 'sweet spot' is and needs extra force to get past the sticking point”

I wonder if this is a mechanical problem with your trimmer

tspear, 2022-03-20:
> “ The trimmer pot sticks when reaching about 2/3 CCW which is roughly where the 'sweet spot' is and needs extra force to get past the sticking point”
> 
> 
> 
> I wonder if this is a mechanical problem with your trimmer 

Yes I might try replacing that first.