TomWhitwell/TuringMachine #175: Turing machine MK2 half working

MattCufre, 2022-04-03:
Hi,

I put together a turing machine over the weekend and it's...half working? I had it working for long enough to calibrate it, so I know it does work, but I moved it to add in the pulses expander and since I've put it back (with or without pulses) it's now not working as expected. The first four leds run as expected, but the second 4 leds are either statically lit, or off. If I use the write switch to remove bits and lock the sequence, it will run for the first 4 leds, but then stop. 

I found that if I put pressure on the power socket (not something I meant to do) that it will then run as expected. Basically I wondered if I'd plugged the power in correctly when it first wasn't working properly, pushed a little harder than maybe I should, it started working, I released, back to half working.

I was wondering if there was a way of narrowing down this to either a chip, my soldering (reasonably experienced with DIY builds, have 10 other builds working as expected), maybe burning a trace (can't see anything obvious), or if it's something I've not thought of. I can post some pictures later if needed. I've also run through the basic troubleshooting steps of checking solder joints, double checking resistor values against build documents and everything seems to be correct.

Thanks in advance for any help!

Thonk-Steve, 2022-04-05:
Hi,

It sounds like either you have a bad power cable or you could do with
giving the power socket a re-solder just to make sure there are no bad
joints.
That's what i'd try next, if it doesn't work after that test for shorts on
power header with a digital multimeter -
https://www.youtube.com/watch?v=qS0SoIiiQCo

If the multimeter test is ok send us photos of both sides of both PCBs.

Steve
Thonk

MattCufre, 2022-04-07:
Thanks for the reply, got it working. 

I had been looking at the chips (as I thought that was the easiest thing to change), and so I swapped a couple over (the ones that there are 2 of) and it suddenly worked without question. I wonder if maybe something wasn't quite seated right in the socket. 

Even though I got it working by guessing at a potential problem, I did resolder the power socket anyway as that seemed a really sensible suggestion.

closed 2022-04-07 by MattCufre