TomWhitwell/TuringMachine #178: Main output produces negative voltages

soundsbrown, 2022-06-29:
I just finished building the TM mk II (bought from Exploding Shed). It looks like it's _mostly_ in good shape:
- the LEDs blink in a plausible fashion when I feed it a square wave LFO
- the calibration procedure appeared to go smoothly
- the pulse and noise jacks produce the output I expect
- the pattern randomizes and locks as expected when I turn the main knob
- the switch clears and sets bits when engaged
- the scale knob attenuates the CV output

The bad news is that there's something seriously off about the main CV output:
- the pattern looks less random than expected, although it's not entirely deterministic (see https://youtu.be/hMXqbaPOg3U)
- it only outputs negative voltages (between 0 and approximately -8.7V, with a lot of time spent stuck to the latter)

In case it matters, the noise output's range is approximately +/-3.5V, and the pulses are +8.3V.

I haven't been able to spot any obvious problems with my soldering work yet, but I'll keep looking. Meanwhile, however, any tips on where I should focus my attention would be very much appreciated!

Thanks,
Tom

soundsbrown, 2022-06-29:
Addendum: I just built the Volts expander as well, and that works as expected as long as I _don't_ install the TL072 on it 😖  Once the op amp is in place, the output range changes to between 8.7V and 9.5V, with occasional temporary dips to 0V. Presumably/hopefully this is another symptom of my problem with the main module 🤞

TomWhitwell, 2022-06-29:
Hi Tom, 
Sorry you're having trouble. Those are extremely odd symptoms. 
Your first comment re negative output points to something wrong either in the DAC0800 or the op amp that scales the output from the DAC (pins 1,2,3 of the TL074 on the front board) - a missing solder point, folded down leg or faulty chip could just about do it. That said, I don't remember ever hearing about negative output before. 
Then your second comment confounds that - the Volts output has no relation to the DAC, just the data stream. So that's very odd. Volts issues are pretty rare. 
So... 
- check all the obvious things - missing joints, reversed chips 
- If you've connected up something weird like power to the expansion header that can kill chips and create odd symptoms 
- Check you haven't confused the 78l09s with the 2n3904 (although you have noise so that's not it) 
- feel free to share hires photos of front & back of both boards here 
- you could try removing the 2x4050 and 2x4081 chips from the back board - they only drive expanders so if there is an error there, the main module should work without them.