TomWhitwell/TuringMachine #181: Length 6 doesn't lock properly

Brokenbar, 2022-08-28:
Hi

I just finished this build, and it is my first build, so there could be something obvious I'm missing. Every other setting works fine, but for length 6, going full ccw on the lock/random-knob means almost all 1-s (there'll be an odd 0 every once in a while) and going full cw is kind of random, but with a tendency towards extremes (mainly towards almost all 0-s with the odd 1, but suddenly it spits out all 1-s for a cycle).

I tried following the schematic to look at the specific soldering points for length 6, but I wasn't able to find anything that looked bad (but I really don't know how to read these things). Looking at out in a scope, I also notice that with length 6 and lock at 12 (+- a couple of "hours"), the signal is jittery within steps, i.e., the main thing where it jumps depending on the bits is still doing its thing, and it seems random-ish, but the signal "at rest" between each clock pulse jumps between two values that depend on scale (attached picture of scope. The clock in is stopped here and scale is at 12, which seems to mean ~half a volt). Does anyone have any ideas what could be causing this or where I should be looking?

[image]

TomWhitwell, 2022-08-28:
Hi, sorry you're having trouble. First place I'd look is the solder points on the back of the rotary switch, then check all over for a missing or grey/dry solder point

Brokenbar, 2022-08-28:
Hm, I tried looking over everything, and I can't find anything of note -- probably because I'm not competent enough. I tried resoldering a couple of the joints I thought looked the most "suspicious" (though none really seemed different from the others in colour, but there were a couple that had a slightly more "bulbous" appearance), but it didn't make any difference. I think maybe since this is my first build, and I'm not capable of focusing my troubleshooting, I might just settle for 6 not working properly and not risk making it worse by resoldering everything. Resoldering seems to be way more difficult than just soldering. Thanks for the help, though! This thing is great fun even without that one length setting working perfectly.