TomWhitwell/TuringMachine #185: Calibration failure, tone interrupts, no sustained tone

Ronny-08, 2022-11-12:
I have assembled the module the last few days and now have problems with calibration.
I run them as in the video, but I don't get no pure sustained constant tone!
The trimmer is set complete CW but but the sound always has short pauses/interruptions.
when i turn the trimmer ccw, the pauses/interruptions get longer until there is no sound at all.
When no sound is heard, the LED's light up permanently.
Could someone give me a tip where to look?

THANKS A LOT

Ronny

Ronny-08, 2022-11-24:
Uploading 69101103858__6984A699-BA58-4154-A6D3-EDB3FA005AB4.MOV…

Ronny-08, 2022-11-24:
https://user-images.githubusercontent.com/118045259/203856149-e36a5a27-8998-42d9-8b3d-64fedaa88b40.mov

marksumptereverton, 2023-03-13:
I've got the exact same issue by the look of things, I wonder did you ever manage to fix this?

Ronny-08, 2023-03-16:
I still have the problem and have not been able to fix the error.
Does anyone have a clue to start troubleshooting.

marksumptereverton, 2023-03-23:
I'm ordering another kit from Thonk, hopefully  this one will work correctly when built up. I'm going to then swap the boards around between working and none working Turing machine ,  I'm hoping this will at least narrow down to which board is faulty. If I have any joy with it I'll let you know Ronny.

Ronny-08, 2023-03-24:
That sounds good!
Thanks already