TomWhitwell/TuringMachine #187: Any advice for sourcing DAC0800LCN

phpguru, 2022-11-20:
Hey Tom, 

Looks like I can get every IC needed for a build except for the DAC. Mouser says more are expected 2024! I read your BOM about cheap Chinese knockoffs and would like to know if you have any other recommendations where to source this IC. Digikey was also a no go. Since I bought a panel pcb only, do you happen to know if Thonk will sell me just the parts bag? 

Also as a long time GitHub user, if an issue is solved you should close it. Do you want help managing issues?

Cheers, Geoff

Thonk-Steve, 2022-11-20:
Hi Geoff, drop us a line on ***@***.*** and we can help you out 👍

phpguru, 2022-11-20:
Thanks Steve, I’m guessing it’s the Thonk support email?

deepiain, 2022-11-20:
I’ve built 100’s of projects and I’ve never had any issues buying chips from eBay and Ali express.. not sure it’s as much of a minefield as people make out Sent from my iPhone

TomWhitwell, 2022-11-20:
@phpguru Yes, it will be support@thonk.co.uk - not sure why the address was obfuscated
And yes, help managing issues would be great - I'm not a very confident GitHub user

closed 2022-11-20 by TomWhitwell