TomWhitwell/TuringMachine #189: Turing Machine

Kolibrix2, 2022-12-10:
Hello I hope you are well. Im buying the kit from the tuning machine last week and now I build it full finish but when I go frame the trigger ( midi cv trigger from Behringer) The Turing machine does nothing ( I go frome trig into clock and then from out into trig of the synt ) if I start the model only 2 LED are one. Kann every one help me to fix that problem ?

Kolibrix2, 2022-12-11:
Good morning I have an update when I connect CV out to Pulse this LED lights up but when I connect it to the clock nothing happens generally does not come out of the out at the top only the two left LEDs and the very right one can possibly someone help thanks in advance

Thonk-Steve, 2022-12-11:
Hi, could you maybe upload a video of what you are describing?

Kolibrix2, 2022-12-11:
I don't know how to go about troubleshooting ( it's my first DIY module ) Here is the video I hope you can help me :)
I hope a YouTube Link its okay : https://www.youtube.com/shorts/lD_nOBUHOGA

Thonk-Steve, 2022-12-11:
Thanks, what is the jack you are patching into on the Behringer called?

Thonk-Steve, 2022-12-11:
> I tried it with CV out and Gate out of the TD3 with my Behringer Midi CV I
> tried the outputs CV and Trig (if I put it directly into my voice it works
> too)

Yes the Pulse jack on the Turing is an OUTPUT, not input. You won't have
damaged anything doing this I don't think though.

Kolibrix2, 2022-12-11:
So I can't trigger the module via the trig out ore CV out of my midi CV from Behringer ?( sorry I'm absolutely new in this world :D )
[image]

Kolibrix2, 2022-12-11:
maybe I need this one ?[ ](https://www.thonk.co.uk/shop/startup/)

Kolibrix2, 2022-12-11:
maybe I need this one ? https://www.thonk.co.uk/shop/startup/

Thonk-Steve, 2022-12-12:
> So I can't trigger the module via the trig out ore CV out of my midi CV
from Behringer ?( sorry I'm absolutely new in this world :D )

No CV out will be sending the note CVS, i.e. the melody it's playing, so
search your Behringer manual for sending the clock out or sync out, i'm not
familiar with that device sorry.

Kolibrix2, 2022-12-16:
Hello, today I tried it with a doepfer a190-8 and unfortunately it didn't work either, maybe someone gave me a tip ? 
[image]
[image]
[image]
[image]
Here are a few pictures, maybe someone recognizes yes, because mistakes, thank you

TomWhitwell, 2022-12-16:
Hi, I'm not certain, but it looks like this chip is the wrong way round. Pin 1 is top left, and pin 1 may be marked by that flattened circle.  
[image]
If your problem is you have no LEDs at all, and no response to the clock out of the Doepfer, that chip wouldn't make the difference, but it could possibly be scrambling / shorting power in the whole module (or have already damaged other chips). Try removing that chip to see if it makes any difference - the looping won't work, but you might at least be able to see pulses with clock input.

Kolibrix2, 2022-12-17:
Thanks for everything, I installed the chip correctly and now it goes out when I go over the pulse but unfortunately nothing comes out of the out output, what could be the reason?

Kolibrix2, 2022-12-17:
Okaay Thanks vor everting