TomWhitwell/TuringMachine #192: CV input expected/safe voltage range?

shorn465, 2023-02-01:
Hi! I just built my Turing Machine, and it’s working beautifully. I’ve been looking all over the web trying to find a place where the expected voltage range for the CV input is documented but I haven’t come across anything. I’m not well versed in reading schematics so if it’s possible to determine that from the schematic I am not able to glean it. I’m interested in knowing:
-Can it accept negative voltage?
-Can it accept voltages above/below 5V/-5V? Is it safe to go up to 10V?
-Does the input voltage add to the knob position, or does the knob become an attenuator/verter for the input voltage?

I can certainly experiment with it but I wanted to know the exact numbers because the nature of not having a sequence fully locked means potentially losing a melody mid performance so I want to know exactly where I have the knob set based on certain voltages. 

Thanks for your help!

TomWhitwell, 2023-02-01:
Hi,

Great to hear from you

Safe range = +/- power supply rails = +/-12v

The voltage adds or subtracts from the knob - so if you put the knob at
fully clockwise / locked, you need to pulse in a negative voltage to unlock
it.

In that example, you'd need about -5v to unlock -- to do the equivalent of
moving the knob to the centre. Similarly if it was fully anticlockwise, +5
should unlock it.

Hope that helps,

Tom

shorn465, 2023-02-02:
Thank you very much! That is exactly what I was looking for.