TomWhitwell/TuringMachine #193: Issue First Start Turing Machine

ulimakalu, 2023-02-06:
I have completed my TM. After switching on for the first time, all the top 8 LEDs are permanently on and unfortunately nothing else happens. A clock is also not received. I can solder and have double checked all solder joints and components. Can someone please give me some advice on where to start troubleshooting? Thanks very much!

Thonk-Steve, 2023-02-06:
Hi, uploading photos of both sides of both PCBs is a good place to start.

ulimakalu, 2023-02-06:
Thank you!
[image]

ulimakalu, 2023-02-06:
[image]

ulimakalu, 2023-02-06:
[image]
[image]
[image]
[image]

ulimakalu, 2023-02-09:
Can someone please give me a link to download the circuit? Is this possible? Thanks very much!

Thonk-Steve, 2023-02-09:
Schematics are here -
https://github.com/TomWhitwell/TuringMachine/blob/master/Collateral/PDF%20Schematics/TuringMachine2_Schematic_May2016.pdf

ulimakalu, 2023-02-10:
Thank you, maybe I can find the mistake.