TomWhitwell/TuringMachine #194: One pulse plays twice with varying strength on second "ghost" pulse.

electricdawn, 2023-02-08:
This is a bit hard to describe, but I'll give it a go. I hope anyone can help me here. Thank you!

So, if I have only one pulse loaded and running circular in the Turing Machine, it will trigger the Pulse output or the outputs on the Pulses expander as intended.

The problem is, that it will trigger a "ghost" pulse, in time with the clock, right after the "real" pulse that varies in strength. Almost like an LFO, but a bit too random for that. I was thinking that it might be a crosstalk problem, maybe caused by me during soldering.

I'd be really glad and thankful if someone could give me some advice on this and where to start testing. Thank you again for your help.

-e

jstenroos, 2023-02-08:
Hi,

Check this thread, it might be helpful: https://github.com/TomWhitwell/TuringMachine/issues/102 <https://github.com/TomWhitwell/TuringMachine/issues/102>

-Jouni

electricdawn, 2023-02-08:
Thank you very much for the quick reply, Jouni!

I'll be looking into the caps solution. Too bad that this is an inherent problem with the design and not something that can be "easily" fixed.

Again, thanks a lot for your help. I really appreciate that. Cheers.

-e

closed 2023-02-08 by electricdawn

jstenroos, 2023-02-08:
The dropbox links are dead in the original thread, here is the image again:

electricdawn, 2023-02-08:
Unfortunately I can't see any image there. But no problem, I should be able to figure this out on my own. :) Thanks again.

-e

jstenroos, 2023-02-08:
[image]

Trying as jpg.

electricdawn, 2023-02-08:
Thank you, Jouni!

That one came through. The pulse of the AND gate just would need to be lengthened a teensy bit. Maybe a small cap there would work as well? Thank you again!

-e